osrf/ros2_test_cases #483: Check action Cli

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 action info /fibonacci
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Action: /fibonacci
       Action clients: 0
       Action servers: 1
         /minimal_action_server
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 action list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /fibonacci
       ```
  </details>
- [ ] **Check send_goal**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci order:\ 2            terminal: 2
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Waiting for an action server to become available...
       Sending goal:
           order: 2
       
       Goal accepted with ID: 0110dbd59e804374bff8e4900c3201e8
       
       Result:
           sequence:
       - 0
       - 1
       - 1
       
       Goal finished with status: SUCCEEDED
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
```
❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp
```

- info
```
❯ ros2 action info /fibonacci
Action: /fibonacci
Action clients: 0
Action servers: 1
    /minimal_action_server
```
- list
```
❯ ros2 action list
/fibonacci
```
- send goal
```
❯ ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci order:\ 2
Waiting for an action server to become available...
Sending goal:
     order: 2

Goal accepted with ID: 12eeec7417884e2f86340a6591fd087e

Result:
    sequence:
- 0
- 1
- 1

Goal finished with status: SUCCEEDED
```

clalancette, 2023-05-07:
Thank you for testing!

closed 2023-05-07 by clalancette