osrf/ros2_test_cases #484: Check action Cli

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 action info /fibonacci
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Action: /fibonacci
       Action clients: 0
       Action servers: 1
         /minimal_action_server
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 action list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /fibonacci
       ```
  </details>
- [ ] **Check send_goal**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci order:\ 2            terminal: 2
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Waiting for an action server to become available...
       Sending goal:
           order: 2
       
       Goal accepted with ID: 0110dbd59e804374bff8e4900c3201e8
       
       Result:
           sequence:
       - 0
       - 1
       - 1
       
       Goal finished with status: SUCCEEDED
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

bai-ivan, 2023-05-06:
Working on this

bai-ivan, 2023-05-06:
1)
```
C:\Windows\System32>ros2 action info /fibonacci
Action: /fibonacci
Action clients: 0
Action servers: 1
    /minimal_action_server
```

2)
```
C:\Windows\System32>ros2 action list
/fibonacci
```
3) issue text got messed up here, but i used the correct one ;) 
```
C:\Windows\System32>ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci "{order: 2}"
Waiting for an action server to become available...
Sending goal:
     order: 2

Goal accepted with ID: b556ba839dcf4bacb223c00e7e0cac2a

Result:
    sequence:
- 0
- 1
- 1

Goal finished with status: SUCCEEDED
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette