osrf/ros2_test_cases #487: Check Lifecyle node py

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Lifecycle Py](https://github.com/ros2/demos/tree/humble/lifecycle_py)

## Checks
- [ ] **Check `listener` and `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run lifecycle_py lifecycle_talker
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 lifecycle nodes
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 lifecycle set /lc_talker 1
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 lifecycle set /lc_talker 3
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [INFO] [1680583079.479073069] [lc_talker]: on_configure() is called.
       [INFO] [1680583080.439717509] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
       ```
    1.
       ```bash
       # stdout in terminal 2
       /lc_talker
       ```
    1.
       ```bash
       # stdout in terminal 2
       Transitioning successful
       ```
    1.
       ```bash
       # stdout in terminal 2
       [INFO] [1680583403.113550155] [lc_talker]: on_activate() is called.
       [INFO] [1680583403.439875193] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #323]
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
terminal 1
```
❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 run lifecycle_py lifecycle_talker
[INFO] [1683388629.220091881] [lc_talker]: on_configure() is called.
[INFO] [1683388630.211646261] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388631.211667579] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388632.211652381] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388633.211653406] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388634.211623274] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388635.211670870] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388636.211973700] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388637.211751100] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388638.211647049] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388639.211594452] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388640.212186929] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388641.070025445] [lc_talker]: on_activate() is called.
[INFO] [1683388641.211703125] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #11]
[INFO] [1683388642.211608131] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #12]
[INFO] [1683388643.211618582] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #13]
[INFO] [1683388644.211640055] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #14]
[INFO] [1683388645.211664026] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #15]
[INFO] [1683388646.211576709] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #16]
```

terminal 2
```
❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 run lifecycle_py lifecycle_talker
[INFO] [1683388629.220091881] [lc_talker]: on_configure() is called.
[INFO] [1683388630.211646261] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388631.211667579] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388632.211652381] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388633.211653406] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388634.211623274] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388635.211670870] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388636.211973700] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388637.211751100] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388638.211647049] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388639.211594452] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388640.212186929] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683388641.070025445] [lc_talker]: on_activate() is called.
[INFO] [1683388641.211703125] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #11]
[INFO] [1683388642.211608131] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #12]
[INFO] [1683388643.211618582] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #13]
[INFO] [1683388644.211640055] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #14]
[INFO] [1683388645.211664026] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #15]
[INFO] [1683388646.211576709] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #16]
```

Yadunund, 2023-05-08:
Thanks for testing!

closed 2023-05-08 by Yadunund