osrf/ros2_test_cases #488: Check Lifecyle node py

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Lifecycle Py](https://github.com/ros2/demos/tree/humble/lifecycle_py)

## Checks
- [ ] **Check `listener` and `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run lifecycle_py lifecycle_talker
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 lifecycle nodes
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 lifecycle set /lc_talker 1
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 lifecycle set /lc_talker 3
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [INFO] [1680583079.479073069] [lc_talker]: on_configure() is called.
       [INFO] [1680583080.439717509] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
       ```
    1.
       ```bash
       # stdout in terminal 2
       /lc_talker
       ```
    1.
       ```bash
       # stdout in terminal 2
       Transitioning successful
       ```
    1.
       ```bash
       # stdout in terminal 2
       [INFO] [1680583403.113550155] [lc_talker]: on_activate() is called.
       [INFO] [1680583403.439875193] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #323]
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

bai-ivan, 2023-05-06:
Working on this

bai-ivan, 2023-05-06:
1) 
```
:\Users\user\ros2_ws>ros2 run lifecycle_py lifecycle_talker
[INFO] [1683357349.953249100] [lc_talker]: on_configure() is called.
```
2)
```
C:\Windows\System32>ros2 lifecycle nodes
/lc_talker
```
3)
```
C:\Windows\System32>ros2 lifecycle set /lc_talker 1
Transitioning successful

C:\Windows\System32>ros2 lifecycle set /lc_talker 3
Transitioning successful
```
4)
```
C:\Users\user\ros2_ws>ros2 run lifecycle_py lifecycle_talker
[INFO] [1683357349.953249100] [lc_talker]: on_configure() is called.
[INFO] [1683357350.891659800] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683357351.889908400] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683357352.892075700] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683357353.884417400] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683357354.889045400] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
[INFO] [1683357355.151561100] [lc_talker]: on_activate() is called.
[INFO] [1683357355.881912100] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #5]
[INFO] [1683357356.884065400] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #6]
[INFO] [1683357357.888067200] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #7]
[INFO] [1683357358.891113700] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #8]
[INFO] [1683357359.892576700] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #9]
[INFO] [1683357360.888988400] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #10]
[INFO] [1683357361.893837500] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #11]
[INFO] [1683357362.894830100] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #12]
[INFO] [1683357363.880781700] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #13]
[INFO] [1683357364.894042000] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #14]
[INFO] [1683357365.891431300] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #15]
[INFO] [1683357366.892752200] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #16]
[INFO] [1683357367.884559200] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #17]
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette