osrf/ros2_test_cases #489: Check Lifecyle node py

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Lifecycle Py](https://github.com/ros2/demos/tree/humble/lifecycle_py)

## Checks
- [ ] **Check `listener` and `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run lifecycle_py lifecycle_talker
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 lifecycle nodes
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 lifecycle set /lc_talker 1
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 lifecycle set /lc_talker 3
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [INFO] [1680583079.479073069] [lc_talker]: on_configure() is called.
       [INFO] [1680583080.439717509] [lc_talker]: Lifecycle publisher is inactive. Messages are not published.
       ```
    1.
       ```bash
       # stdout in terminal 2
       /lc_talker
       ```
    1.
       ```bash
       # stdout in terminal 2
       Transitioning successful
       ```
    1.
       ```bash
       # stdout in terminal 2
       [INFO] [1680583403.113550155] [lc_talker]: on_activate() is called.
       [INFO] [1680583403.439875193] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #323]
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
[image]

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette