osrf/ros2_test_cases #490: Check ROS 2 bag CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check record**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You should stop the talker once the recording is stopped.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
    1.
       Run for a few seconds and then hit Ctrl-C to stop the bag recording.
       ```bash
       # User input in terminal 2
       ros2 bag record /chatter -o my-bag
       ```
  
    **Expect**
    
    1.
       You should see a folder named my-bag in the current directory
    
  </details>
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Continuing from the previous check.
       
    1.
       ```bash
       # User input in terminal 1
       ros2 bag info my-bag
       ```
  
    **Expect**
    
    1.
       Something like the following
       ```bash
       # stdout in terminal 1
       Files:             my-bag_0.mcap
       Bag size:          3.0 KiB
       Storage id:        mcap
       Duration:          3.999s
       Start:             Apr  2 2023 13:45:47.834 (1680414347.834)
       End:               Apr  2 2023 13:45:51.834 (1680414351.834)
       Messages:          5
       Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 5 | Serialization Format: cdr
       ```
  </details>
- [ ] **Check play**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Continuing from the previous two checks. Make sure that the talker is stopped.
       
    1.
       The `-l` option loops the bag. You should kill this process eventually with Ctrl-C.
       ```bash
       # User input in terminal 1
       ros2 bag play -l my-bag
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic echo /chatter
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       ---
       data: 'Hello World: 77'
       ---
       data: 'Hello World: 78'
       ---
       data: 'Hello World: 75'
       ---
       data: 'Hello World: 76'
       ---
       data: 'Hello World: 77'
       ---
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
**The bag CLI is working as expected. All the checks are passed.**

Yadunund, 2023-05-09:
Thanks for testing!