osrf/ros2_test_cases #491: Check ROS 2 bag CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check record**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You should stop the talker once the recording is stopped.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
    1.
       Run for a few seconds and then hit Ctrl-C to stop the bag recording.
       ```bash
       # User input in terminal 2
       ros2 bag record /chatter -o my-bag
       ```
  
    **Expect**
    
    1.
       You should see a folder named my-bag in the current directory
    
  </details>
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Continuing from the previous check.
       
    1.
       ```bash
       # User input in terminal 1
       ros2 bag info my-bag
       ```
  
    **Expect**
    
    1.
       Something like the following
       ```bash
       # stdout in terminal 1
       Files:             my-bag_0.mcap
       Bag size:          3.0 KiB
       Storage id:        mcap
       Duration:          3.999s
       Start:             Apr  2 2023 13:45:47.834 (1680414347.834)
       End:               Apr  2 2023 13:45:51.834 (1680414351.834)
       Messages:          5
       Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 5 | Serialization Format: cdr
       ```
  </details>
- [ ] **Check play**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Continuing from the previous two checks. Make sure that the talker is stopped.
       
    1.
       The `-l` option loops the bag. You should kill this process eventually with Ctrl-C.
       ```bash
       # User input in terminal 1
       ros2 bag play -l my-bag
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic echo /chatter
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       ---
       data: 'Hello World: 77'
       ---
       data: 'Hello World: 78'
       ---
       data: 'Hello World: 75'
       ---
       data: 'Hello World: 76'
       ---
       data: 'Hello World: 77'
       ---
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
- record
```
❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 bag record /chatter -o my-bag
[INFO] [1683388793.673052097] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1683388793.675286062] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1683388793.675342525] [rosbag2_recorder]: Listening for topics...
[INFO] [1683388793.676583195] [rosbag2_recorder]: Subscribed to topic '/chatter'
[INFO] [1683388793.676729731] [rosbag2_recorder]: All requested topics are subscribed. Stopping discovery...
[INFO] [1683388798.010881994] [rclcpp]: signal_handler(signum=2)
[INFO] [1683388798.011191213] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1683388798.046345136] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while

~
❯ ls my-bag
metadata.yaml  my-bag_0.mcap
```
- info
```
❯ ros2 bag info my-bag

Files:             my-bag_0.mcap
Bag size:          3.2 KiB
Storage id:        mcap
Duration:          2.999s
Start:             May  7 2023 00:59:54.502 (1683388794.502)
End:               May  7 2023 00:59:57.502 (1683388797.502)
Messages:          4
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 4 | Serialization Format: cdr
```
- play
```
❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 bag play -l my-bag
[INFO] [1683388930.895429473] [rosbag2_player]: Set rate to 1
[INFO] [1683388930.896811767] [rosbag2_player]: Adding keyboard callbacks.
[INFO] [1683388930.896828486] [rosbag2_player]: Press SPACE for Pause/Resume
[INFO] [1683388930.896834400] [rosbag2_player]: Press CURSOR_RIGHT for Play Next Message
[INFO] [1683388930.896839603] [rosbag2_player]: Press CURSOR_UP for Increase Rate 10%
[INFO] [1683388930.896844531] [rosbag2_player]: Press CURSOR_DOWN for Decrease Rate 10%
[INFO] [1683388930.896895640] [rosbag2_player]: Playback until timestamp: -1

❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 topic echo /chatter
data: 'Hello World: 6'
---
data: 'Hello World: 7'
---
data: 'Hello World: 4'
---
data: 'Hello World: 5'
---
data: 'Hello World: 6'
---
data: 'Hello World: 7'
---
data: 'Hello World: 4'
```

Yadunund, 2023-05-08:
Thank you for testing!

closed 2023-05-08 by Yadunund