osrf/ros2_test_cases #492: Check ROS 2 bag CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check record**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You should stop the talker once the recording is stopped.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
    1.
       Run for a few seconds and then hit Ctrl-C to stop the bag recording.
       ```bash
       # User input in terminal 2
       ros2 bag record /chatter -o my-bag
       ```
  
    **Expect**
    
    1.
       You should see a folder named my-bag in the current directory
    
  </details>
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Continuing from the previous check.
       
    1.
       ```bash
       # User input in terminal 1
       ros2 bag info my-bag
       ```
  
    **Expect**
    
    1.
       Something like the following
       ```bash
       # stdout in terminal 1
       Files:             my-bag_0.mcap
       Bag size:          3.0 KiB
       Storage id:        mcap
       Duration:          3.999s
       Start:             Apr  2 2023 13:45:47.834 (1680414347.834)
       End:               Apr  2 2023 13:45:51.834 (1680414351.834)
       Messages:          5
       Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 5 | Serialization Format: cdr
       ```
  </details>
- [ ] **Check play**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Continuing from the previous two checks. Make sure that the talker is stopped.
       
    1.
       The `-l` option loops the bag. You should kill this process eventually with Ctrl-C.
       ```bash
       # User input in terminal 1
       ros2 bag play -l my-bag
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic echo /chatter
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       ---
       data: 'Hello World: 77'
       ---
       data: 'Hello World: 78'
       ---
       data: 'Hello World: 75'
       ---
       data: 'Hello World: 76'
       ---
       data: 'Hello World: 77'
       ---
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

bai-ivan, 2023-05-06:
Working on this

bai-ivan, 2023-05-06:
1) 
```
C:\Windows\System32>cd my-bag

C:\Windows\System32\my-bag>dir
 Volume in drive C has no label.
 Volume Serial Number is 2419-887D

 Directory of C:\Windows\System32\my-bag

06.05.2023 г.  10:19    <DIR>          .
06.05.2023 г.  10:19    <DIR>          ..
06.05.2023 г.  10:19             1 114 metadata.yaml
06.05.2023 г.  10:19             2 772 my-bag_0.mcap
```
2)
```
c:\Windows\System32>ros2 bag info my-bag

Files:             my-bag_0.mcap
Bag size:          3.8 KiB
Storage id:        mcap
Duration:          11.6s
Start:             May  6 2023 10:19:09.45 (1683357549.45)
End:               May  6 2023 10:19:20.51 (1683357560.51)
Messages:          12
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 12 | Serialization Format: cdr
```
3)
terminal 1
```
c:\Windows\System32>ros2 bag play -l my-bag
[INFO] [1683357716.179258200] [rosbag2_player]: Set rate to 1
[WARN] [1683357716.195060300] [rclcpp]: A zero depth with KEEP_LAST doesn't make sense; no data could be stored.This will be interpreted as SYSTEM_DEFAULT
[INFO] [1683357716.208626800] [rosbag2_player]: Adding keyboard callbacks.
[INFO] [1683357716.220452500] [rosbag2_player]: Press SPACE for Pause/Resume
[INFO] [1683357716.221563900] [rosbag2_player]: Press CURSOR_RIGHT for Play Next Message
[INFO] [1683357716.226164100] [rosbag2_player]: Press CURSOR_UP for Increase Rate 10%
[INFO] [1683357716.233880700] [rosbag2_player]: Press CURSOR_DOWN for Decrease Rate 10%
[INFO] [1683357716.243940400] [rosbag2_player]: Playback until timestamp: -1
```

terminal 2
```
C:\Windows\System32\my-bag>ros2 topic echo /chatter
data: 'Hello World: 17'
---
data: 'Hello World: 18'
---
data: 'Hello World: 19'
---
data: 'Hello World: 20'
---
data: 'Hello World: 21'
---
data: 'Hello World: 10'
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette