osrf/ros2_test_cases #493: Check ROS 2 bag CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check record**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You should stop the talker once the recording is stopped.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
    1.
       Run for a few seconds and then hit Ctrl-C to stop the bag recording.
       ```bash
       # User input in terminal 2
       ros2 bag record /chatter -o my-bag
       ```
  
    **Expect**
    
    1.
       You should see a folder named my-bag in the current directory
    
  </details>
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Continuing from the previous check.
       
    1.
       ```bash
       # User input in terminal 1
       ros2 bag info my-bag
       ```
  
    **Expect**
    
    1.
       Something like the following
       ```bash
       # stdout in terminal 1
       Files:             my-bag_0.mcap
       Bag size:          3.0 KiB
       Storage id:        mcap
       Duration:          3.999s
       Start:             Apr  2 2023 13:45:47.834 (1680414347.834)
       End:               Apr  2 2023 13:45:51.834 (1680414351.834)
       Messages:          5
       Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 5 | Serialization Format: cdr
       ```
  </details>
- [ ] **Check play**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Continuing from the previous two checks. Make sure that the talker is stopped.
       
    1.
       The `-l` option loops the bag. You should kill this process eventually with Ctrl-C.
       ```bash
       # User input in terminal 1
       ros2 bag play -l my-bag
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic echo /chatter
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       ---
       data: 'Hello World: 77'
       ---
       data: 'Hello World: 78'
       ---
       data: 'Hello World: 75'
       ---
       data: 'Hello World: 76'
       ---
       data: 'Hello World: 77'
       ---
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
```
❯ ros2 bag record /chatter -o my-bag
[INFO] [1683346307.823767360] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1683346307.825885134] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1683346307.825935970] [rosbag2_recorder]: Listening for topics...
[INFO] [1683346307.826989975] [rosbag2_recorder]: Subscribed to topic '/chatter'
[INFO] [1683346307.827140429] [rosbag2_recorder]: All requested topics are subscribed. Stopping discovery...
[INFO] [1683346314.437917986] [rclcpp]: signal_handler(signum=2)
[INFO] [1683346314.438595483] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1683346314.496964375] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
```
```
❯ ls my-bag
metadata.yaml  my-bag_0.mcap
```
```
❯ ros2 bag info my-bag

Files:             my-bag_0.mcap
Bag size:          3.4 KiB
Storage id:        mcap
Duration:          5.999s
Start:             May  6 2023 13:11:48.144 (1683346308.144)
End:               May  6 2023 13:11:54.144 (1683346314.144)
Messages:          7
Topic information: Topic: /chatter | Type: std_msgs/msg/String | Count: 7 | Serialization Format: cdr
```
```
❯ ros2 bag play -l my-bag
[INFO] [1683346451.411415227] [rosbag2_player]: Set rate to 1
[INFO] [1683346451.412841194] [rosbag2_player]: Adding keyboard callbacks.
[INFO] [1683346451.412859306] [rosbag2_player]: Press SPACE for Pause/Resume
[INFO] [1683346451.412865955] [rosbag2_player]: Press CURSOR_RIGHT for Play Next Message
[INFO] [1683346451.412872070] [rosbag2_player]: Press CURSOR_UP for Increase Rate 10%
[INFO] [1683346451.412878030] [rosbag2_player]: Press CURSOR_DOWN for Decrease Rate 10%
[INFO] [1683346451.412930154] [rosbag2_player]: Playback until timestamp: -1
```
```
❯ ros2 topic echo /chatter
data: 'Hello World: 35'
---
data: 'Hello World: 29'
---
data: 'Hello World: 30'
---
data: 'Hello World: 31'
---
data: 'Hello World: 32'
---
data: 'Hello World: 33'
---

```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette