osrf/ros2_test_cases #495: Check ROS 2 component CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rclcpp_components component_container
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 component list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /ComponentManager
       ```
  </details>
- [ ] **Check load**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rclcpp_components component_container
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 component load /ComponentManager action_tutorials_cpp action_tutorials_cpp::FibonacciActionServer
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [INFO] [1680446030.083590920] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<action_tutorials_cpp::FibonacciActionServer>
       [INFO] [1680446030.083610890] [ComponentManager]: Instantiate class: rclcpp_components::NodeFactoryTemplate<action_tutorials_cpp::FibonacciActionServer>
       ```
    1.
       ```bash
       # stdout in terminal 2
       Loaded component 1 into '/ComponentManager' container node as '/fibonacci_action_server'
       ```
  </details>
- [ ] **Check standalone**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rclcpp_components component_container
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 component standalone action_tutorials_cpp action_tutorials_cpp::FibonacciActionServer
       ```
    1.
       ```bash
       # User input in terminal 3
       ros2 component list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       [INFO] [1680446030.083590920] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<action_tutorials_cpp::FibonacciActionServer>
       [INFO] [1680446030.083610890] [ComponentManager]: Instantiate class: rclcpp_components::NodeFactoryTemplate<action_tutorials_cpp::FibonacciActionServer>
       ```
    1.
       ```bash
       # stdout in terminal 3
       /ComponentManager
       /standalone_container_8269218e7494
         1  /fibonacci_action_server
       ```
  </details>
- [ ] **Check types**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 component types
       ```
  
    **Expect**
    
    1.
       Something like the following
       ```bash
       # stdout in terminal 1
       robot_state_publisher
         robot_state_publisher::RobotStatePublisher
       tf2_ros
         tf2_ros::StaticTransformBroadcasterNode
       quality_of_service_demo_cpp
         quality_of_service_demo::MessageLostListener
         quality_of_service_demo::MessageLostTalker
         quality_of_service_demo::QosOverridesListener
         quality_of_service_demo::QosOverridesTalker
       image_tools
         image_tools::Cam2Image
         image_tools::ShowImage
       logging_demo
         logging_demo::LoggerConfig
         logging_demo::LoggerUsage
       examples_rclcpp_wait_set
         Talker
         Listener
       examples_rclcpp_minimal_subscriber
         WaitSetSubscriber
         StaticWaitSetSubscriber
         TimeTriggeredWaitSetSubscriber
       demo_nodes_cpp_native
         demo_nodes_cpp_native::Talker
       demo_nodes_cpp
         demo_nodes_cpp::OneOffTimerNode
         demo_nodes_cpp::ReuseTimerNode
         demo_nodes_cpp::ServerNode
         demo_nodes_cpp::ClientNode
         demo_nodes_cpp::IntrospectionServiceNode
         demo_nodes_cpp::IntrospectionClientNode
         demo_nodes_cpp::ListParameters
         demo_nodes_cpp::ParameterBlackboard
         demo_nodes_cpp::SetAndGetParameters
         demo_nodes_cpp::ParameterEventsAsyncNode
         demo_nodes_cpp::EvenParameterNode
         demo_nodes_cpp::SetParametersCallback
         demo_nodes_cpp::ContentFilteringPublisher
         demo_nodes_cpp::ContentFilteringSubscriber
         demo_nodes_cpp::Talker
         demo_nodes_cpp::LoanedMessageTalker
         demo_nodes_cpp::SerializedMessageTalker
         demo_nodes_cpp::Listener
         demo_nodes_cpp::SerializedMessageListener
         demo_nodes_cpp::ListenerBestEffort
       composition
         composition::Talker
         composition::Listener
         composition::NodeLikeListener
         composition::Server
         composition::Client
       action_tutorials_cpp
         action_tutorials_cpp::FibonacciActionClient
         action_tutorials_cpp::FibonacciActionServer
       ```
  </details>
- [ ] **Check unload**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rclcpp_components component_container
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 component load action_tutorials_cpp action_tutorials_cpp::FibonacciActionServer
       ```
    1.
       ```bash
       # User input in terminal 3
       ros2 component unload /ComponentManager 1
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Loaded component 1 into '/ComponentManager' container node as '/fibonacci_action_server'
       ```
    1.
       ```bash
       # stdout in terminal 3
       Unloaded component 1 from '/ComponentManager' container node
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
- list
```
❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 component list
/ComponentManager
```
- load
```
❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 run rclcpp_components component_container
[INFO] [1683389105.230380842] [ComponentManager]: Load Library: /opt/ros/iron/lib/libaction_tutorials.so
[INFO] [1683389105.233457683] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<action_tutorials_cpp::FibonacciActionClient>
[INFO] [1683389105.233511181] [ComponentManager]: Found class: rclcpp_components::NodeFactoryTemplate<action_tutorials_cpp::FibonacciActionServer>
[INFO] [1683389105.233529603] [ComponentManager]: Instantiate class: rclcpp_components::NodeFactoryTemplate<action_tutorials_cpp::FibonacciActionServer>

❯ ros2 component load /ComponentManager action_tutorials_cpp action_tutorials_cpp::FibonacciActionServer
Loaded component 1 into '/ComponentManager' container node as '/fibonacci_action_server'
```
- standalone
```
❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 component standalone action_tutorials_cpp action_tutorials_cpp::FibonacciActionServer
[INFO] [1683389194.748047230] [standalone_container_aad41ae4cd48]: Load Library: /opt/ros/iron/lib/libaction_tutorials.so
[INFO] [1683389194.748715072] [standalone_container_aad41ae4cd48]: Found class: rclcpp_components::NodeFactoryTemplate<action_tutorials_cpp::FibonacciActionClient>
[INFO] [1683389194.748732039] [standalone_container_aad41ae4cd48]: Found class: rclcpp_components::NodeFactoryTemplate<action_tutorials_cpp::FibonacciActionServer>
[INFO] [1683389194.748737820] [standalone_container_aad41ae4cd48]: Instantiate class: rclcpp_components::NodeFactoryTemplate<action_tutorials_cpp::FibonacciActionServer>

❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 component list
/ComponentManager
/standalone_container_aad41ae4cd48
  1  /fibonacci_action_server
```
- types
```
❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 component types
demo_nodes_cpp_native
  demo_nodes_cpp_native::Talker
robot_state_publisher
  robot_state_publisher::RobotStatePublisher
demo_nodes_cpp
  demo_nodes_cpp::OneOffTimerNode
  demo_nodes_cpp::ReuseTimerNode
  demo_nodes_cpp::ServerNode
  demo_nodes_cpp::ClientNode
  demo_nodes_cpp::IntrospectionServiceNode
  demo_nodes_cpp::IntrospectionClientNode
  demo_nodes_cpp::ListParameters
  demo_nodes_cpp::ParameterBlackboard
  demo_nodes_cpp::SetAndGetParameters
  demo_nodes_cpp::ParameterEventsAsyncNode
  demo_nodes_cpp::EvenParameterNode
  demo_nodes_cpp::SetParametersCallback
  demo_nodes_cpp::ContentFilteringPublisher
  demo_nodes_cpp::ContentFilteringSubscriber
  demo_nodes_cpp::Talker
  demo_nodes_cpp::LoanedMessageTalker
  demo_nodes_cpp::SerializedMessageTalker
  demo_nodes_cpp::Listener
  demo_nodes_cpp::SerializedMessageListener
  demo_nodes_cpp::ListenerBestEffort
composition
  composition::Talker
  composition::Listener
  composition::NodeLikeListener
  composition::Server
  composition::Client
action_tutorials_cpp
  action_tutorials_cpp::FibonacciActionClient
  action_tutorials_cpp::FibonacciActionServer
examples_rclcpp_wait_set
  Talker
  Listener
examples_rclcpp_minimal_subscriber
  WaitSetSubscriber
  StaticWaitSetSubscriber
  TimeTriggeredWaitSetSubscriber
tf2_ros
  tf2_ros::StaticTransformBroadcasterNode
```
- unload
```
❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 component load /ComponentManager action_tutorials_cpp action_tutorials_cpp::FibonacciActionServer
Loaded component 1 into '/ComponentManager' container node as '/fibonacci_action_server'

❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 component unload /ComponentManager 1
Unloaded component 1 from '/ComponentManager' container node
```

Yadunund, 2023-05-08:
Thanks for testing!

closed 2023-05-08 by Yadunund