osrf/ros2_test_cases #500: Check ROS 2 daemon CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check status**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon status
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon is running
       ```
  </details>
- [ ] **Stop the daemon**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon stop
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon has been stopped
       ```
  </details>
- [ ] **Check that the daemon is not running**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon status
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon is not running
       ```
  </details>
- [ ] **Restart the daemon**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon start
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon has been started
       ```
  </details>
- [ ] **Check restart**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon status
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon is running
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

bai-ivan, 2023-05-06:
working on this

bai-ivan, 2023-05-06:
```
c:\Windows\System32>ros2 daemon status
The daemon is running

c:\Windows\System32>ros2 daemon stop
The daemon has been stopped

c:\Windows\System32>ros2 daemon status
The daemon is not running

c:\Windows\System32>ros2 daemon start
The daemon has been started

c:\Windows\System32>ros2 daemon status
The daemon is running
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette