osrf/ros2_test_cases #502: Check ROS 2 doctor CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `-r`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 doctor -r
       ```
  
    **Expect**
    
    1.
       A large report is generated with several sections.
    
  </details>
- [ ] **Check `-rf`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 doctor -rf
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., 'All 5 checks passed').
    
  </details>
- [ ] **Check `-iw`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 doctor -iw
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., '2/5 checks passed').
    
  </details>
- [ ] **Check hello**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 doctor hello
       ```
  
    **Expect**
    
    1.
       You'll see repeated messages that look somewhat like the following.
       ```bash
       # stdout in terminal 1
       MULTIMACHINE COMMUNICATION SUMMARY
       Topic: /canyouhearme, Published Msg Count: 11
       Subscribed from:
                       Hostname             Msg Count /1.0s
       Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
       Received from:
                       Hostname             Msg Count /1.0s
       ------------------------------------------------------------
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
**The doctor CLI works as expected. All checks are passed.**

Additionally: The network python module is not working properly in windows which is found in ```-rf``` and ```-iw```.

Yadunund, 2023-05-09:
Thanks for testing! I commented on the network test here https://github.com/osrf/ros2_test_cases/issues/546#issuecomment-1539770453

closed 2023-05-09 by Yadunund