osrf/ros2_test_cases #511: Check ROS 2 launch CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check launch package**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener.launch.py
       ```
  
    **Expect**
    
    1.
       You should see something like the following
       ```bash
       # stdout in terminal 1
       [INFO] [launch]: All log files can be found below /home/parallels/.ros/log/2022-03-21-17-38-54-737866-parallels-Parallels-Virtual-Platform-3717
       [INFO] [launch]: Default logging verbosity is set to INFO
       [INFO] [talker-1]: process started with pid [3722]
       [INFO] [listener-2]: process started with pid [3724]
       [talker-1] [INFO] [1647909535.937731548] [talker]: Publishing: 'Hello World: 1'
       [listener-2] [INFO] [1647909535.938337387] [listener]: I heard: [Hello World: 1]
       [talker-1] [INFO] [1647909536.937932068] [talker]: Publishing: 'Hello World: 2'
       [listener-2] [INFO] [1647909536.938125486] [listener]: I heard: [Hello World: 2]
       [talker-1] [INFO] [1647909537.936162331] [talker]: Publishing: 'Hello World: 3'
       [listener-2] [INFO] [1647909537.936366391] [listener]: I heard: [Hello World: 3]
       ...
       ```
  </details>
- [ ] **Check launch launch file directly**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Download a launch file, for example with `curl`.
       ```bash
       # User input in terminal 1
       curl https://raw.githubusercontent.com/ros2/demos/master/demo_nodes_cpp/launch/topics/talker_listener.launch.py > /tmp/talker_listener.launch.py
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 launch /tmp/talker_listener.launch.py
       ```
  
    **Expect**
    
    1.
       You should see something like the following
       ```bash
       # stdout in terminal 1
       [INFO] [launch]: All log files can be found below /home/parallels/.ros/log/2022-03-21-17-38-54-737866-parallels-Parallels-Virtual-Platform-3717
       [INFO] [launch]: Default logging verbosity is set to INFO
       [INFO] [talker-1]: process started with pid [3722]
       [INFO] [listener-2]: process started with pid [3724]
       [talker-1] [INFO] [1647909535.937731548] [talker]: Publishing: 'Hello World: 1'
       [listener-2] [INFO] [1647909535.938337387] [listener]: I heard: [Hello World: 1]
       [talker-1] [INFO] [1647909536.937932068] [talker]: Publishing: 'Hello World: 2'
       [listener-2] [INFO] [1647909536.938125486] [listener]: I heard: [Hello World: 2]
       [talker-1] [INFO] [1647909537.936162331] [talker]: Publishing: 'Hello World: 3'
       [listener-2] [INFO] [1647909537.936366391] [listener]: I heard: [Hello World: 3]
       ...
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
```
❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ ros2 launch demo_nodes_cpp talker_listener_launch.py
[INFO] [launch]: All log files can be found below /home/daisuke/.ros/log/2023-05-07-01-20-02-185672-npc2301008-158320
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [158323]
[INFO] [listener-2]: process started with pid [158325]
[talker-1] [INFO] [1683390003.285483157] [talker]: Publishing: 'Hello World: 1'
[listener-2] [INFO] [1683390003.285966289] [listener]: I heard: [Hello World: 1]
[talker-1] [INFO] [1683390004.285521687] [talker]: Publishing: 'Hello World: 2'
[listener-2] [INFO] [1683390004.285679923] [listener]: I heard: [Hello World: 2]
[talker-1] [INFO] [1683390005.285455909] [talker]: Publishing: 'Hello World: 3'
[listener-2] [INFO] [1683390005.285658742] [listener]: I heard: [Hello World: 3]
```
```
❯ ros2 doctor -r | grep middleware
middleware name    : rmw_cyclonedds_cpp

~
❯ curl https://raw.githubusercontent.com/ros2/demos/master/demo_nodes_cpp/launch/topics/talker_listener_launch.py > /tmp/talker_listener_launch.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1001  100  1001    0     0   2647      0 --:--:-- --:--:-- --:--:--  2648

~
❯ ros2 launch /tmp/talker_listener_launch.py
[INFO] [launch]: All log files can be found below /home/daisuke/.ros/log/2023-05-07-01-21-13-100338-npc2301008-158636
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [158639]
[INFO] [listener-2]: process started with pid [158641]
[talker-1] [INFO] [1683390074.206880760] [talker]: Publishing: 'Hello World: 1'
[listener-2] [INFO] [1683390074.207147128] [listener]: I heard: [Hello World: 1]
[talker-1] [INFO] [1683390075.206860770] [talker]: Publishing: 'Hello World: 2'
[listener-2] [INFO] [1683390075.207114807] [listener]: I heard: [Hello World: 2]
[talker-1] [INFO] [1683390076.206873155] [talker]: Publishing: 'Hello World: 3'
[listener-2] [INFO] [1683390076.207166383] [listener]: I heard: [Hello World: 3]
[talker-1] [INFO] [1683390077.206809128] [talker]: Publishing: 'Hello World: 4'
[listener-2] [INFO] [1683390077.207081043] [listener]: I heard: [Hello World: 4]
```

Yadunund, 2023-05-08:
Thanks for testing!

closed 2023-05-08 by Yadunund