osrf/ros2_test_cases #513: Check ROS 2 launch CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check launch package**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener.launch.py
       ```
  
    **Expect**
    
    1.
       You should see something like the following
       ```bash
       # stdout in terminal 1
       [INFO] [launch]: All log files can be found below /home/parallels/.ros/log/2022-03-21-17-38-54-737866-parallels-Parallels-Virtual-Platform-3717
       [INFO] [launch]: Default logging verbosity is set to INFO
       [INFO] [talker-1]: process started with pid [3722]
       [INFO] [listener-2]: process started with pid [3724]
       [talker-1] [INFO] [1647909535.937731548] [talker]: Publishing: 'Hello World: 1'
       [listener-2] [INFO] [1647909535.938337387] [listener]: I heard: [Hello World: 1]
       [talker-1] [INFO] [1647909536.937932068] [talker]: Publishing: 'Hello World: 2'
       [listener-2] [INFO] [1647909536.938125486] [listener]: I heard: [Hello World: 2]
       [talker-1] [INFO] [1647909537.936162331] [talker]: Publishing: 'Hello World: 3'
       [listener-2] [INFO] [1647909537.936366391] [listener]: I heard: [Hello World: 3]
       ...
       ```
  </details>
- [ ] **Check launch launch file directly**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Download a launch file, for example with `curl`.
       ```bash
       # User input in terminal 1
       curl https://raw.githubusercontent.com/ros2/demos/master/demo_nodes_cpp/launch/topics/talker_listener.launch.py > /tmp/talker_listener.launch.py
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 launch /tmp/talker_listener.launch.py
       ```
  
    **Expect**
    
    1.
       You should see something like the following
       ```bash
       # stdout in terminal 1
       [INFO] [launch]: All log files can be found below /home/parallels/.ros/log/2022-03-21-17-38-54-737866-parallels-Parallels-Virtual-Platform-3717
       [INFO] [launch]: Default logging verbosity is set to INFO
       [INFO] [talker-1]: process started with pid [3722]
       [INFO] [listener-2]: process started with pid [3724]
       [talker-1] [INFO] [1647909535.937731548] [talker]: Publishing: 'Hello World: 1'
       [listener-2] [INFO] [1647909535.938337387] [listener]: I heard: [Hello World: 1]
       [talker-1] [INFO] [1647909536.937932068] [talker]: Publishing: 'Hello World: 2'
       [listener-2] [INFO] [1647909536.938125486] [listener]: I heard: [Hello World: 2]
       [talker-1] [INFO] [1647909537.936162331] [talker]: Publishing: 'Hello World: 3'
       [listener-2] [INFO] [1647909537.936366391] [listener]: I heard: [Hello World: 3]
       ...
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
launch package
```
❯ ros2 launch demo_nodes_cpp talker_listener_launch.py
[INFO] [launch]: All log files can be found below /home/daisuke/.ros/log/2023-05-06-13-33-51-915285-npc2301008-23755
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [23758]
[INFO] [listener-2]: process started with pid [23760]
[talker-1] [INFO] [1683347633.028178647] [talker]: Publishing: 'Hello World: 1'
[listener-2] [INFO] [1683347633.028459795] [listener]: I heard: [Hello World: 1]
[talker-1] [INFO] [1683347634.028121265] [talker]: Publishing: 'Hello World: 2'
[listener-2] [INFO] [1683347634.028287285] [listener]: I heard: [Hello World: 2]
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[talker-1] [INFO] [1683347634.853971571] [rclcpp]: signal_handler(signum=2)
[listener-2] [INFO] [1683347634.854064033] [rclcpp]: signal_handler(signum=2)
[INFO] [listener-2]: process has finished cleanly [pid 23760]
[INFO] [talker-1]: process has finished cleanly [pid 23758]
```
Note: I changed `talker_listener.launch.py` to `talker_listener_launch.py`.

wep21, 2023-05-06:
launch launch file directly
```
❯ ros2 launch /tmp/talker_listener_launch.py
[INFO] [launch]: All log files can be found below /home/daisuke/.ros/log/2023-05-06-13-38-57-751437-npc2301008-24140
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [24144]
[INFO] [listener-2]: process started with pid [24146]
[talker-1] [INFO] [1683347938.853503884] [talker]: Publishing: 'Hello World: 1'
[listener-2] [INFO] [1683347938.853945139] [listener]: I heard: [Hello World: 1]
[talker-1] [INFO] [1683347939.853444436] [talker]: Publishing: 'Hello World: 2'
[listener-2] [INFO] [1683347939.853787493] [listener]: I heard: [Hello World: 2]
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[talker-1] [INFO] [1683347940.853476135] [talker]: Publishing: 'Hello World: 3'
[talker-1] [INFO] [1683347940.853567047] [rclcpp]: signal_handler(signum=2)
[listener-2] [INFO] [1683347940.853576008] [rclcpp]: signal_handler(signum=2)
[listener-2] [INFO] [1683347940.853639003] [listener]: I heard: [Hello World: 3]
[INFO] [listener-2]: process has finished cleanly [pid 24146]
[INFO] [talker-1]: process has finished cleanly [pid 24144]
```
Note: I also changed the filename as I showed above.

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette