osrf/ros2_test_cases #516: Check ROS 2 Multicast CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check receive**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 multicast receive
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 multicast send
       ```
  
    
  </details>
- [ ] **Check send**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # stdout in terminal 1
       Waiting for UDP multicast datagram...
       Received from 10.211.55.9:59527: 'Hello World!'
       ```
    1.
       ```bash
       # stdout in terminal 2
       Sending one UDP multicast datagram...
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-06:
On a default Windows 11 binary installation with fastDDS.

Checked the 2 CLI commands for `ros2 multicast`

Commands and behavious are working.

### **_WARNING_**: The testcase itself  is a bit messed up. The `Try` part of   `Check send`  section is rather the `Expectation` from the first set of commands. Should probably only have a single section `Check send and receive`, if possible.

[image]

clalancette, 2023-05-06:
> ### **_WARNING_**: The testcase itself is a bit messed up. The `Try` part of `Check send` section is rather the `Expectation` from the first set of commands. Should probably only have a single section `Check send and receive`, if possible.

Yes, good call.  I've opened https://github.com/audrow/yatm/pull/355 to fix that.

Since the test otherwise passed, I'm going to close this.  Thanks for testing!

closed 2023-05-06 by clalancette