osrf/ros2_test_cases #517: Check ROS 2 Multicast CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check receive**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 multicast receive
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 multicast send
       ```
  
    
  </details>
- [ ] **Check send**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # stdout in terminal 1
       Waiting for UDP multicast datagram...
       Received from 10.211.55.9:59527: 'Hello World!'
       ```
    1.
       ```bash
       # stdout in terminal 2
       Sending one UDP multicast datagram...
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
```
❯ ros2 multicast receive
Waiting for UDP multicast datagram...
Received from 192.168.3.4:47535: 'Hello World!'
```
```
❯ ros2 multicast receive
Waiting for UDP multicast datagram...
Received from 192.168.3.4:47535: 'Hello World!'
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette