osrf/ros2_test_cases #520: Check ROS 2 Node CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Launch a node
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener.launch.py
       ```
    1.
       Get the info for the `talker` that was launched.
       ```bash
       # User input in terminal 2
       ros2 node info /talker
       ```
  
    **Expect**
    
    1.
       You should see something similar to the following
       ```bash
       # stdout in terminal 2
       Subscribers:
         /parameter_events: rcl_interfaces/msg/ParameterEvent
       Publishers:
         /chatter: std_msgs/msg/String
         /parameter_events: rcl_interfaces/msg/ParameterEvent
         /rosout: rcl_interfaces/msg/Log
       Service Servers:
         /talker/describe_parameters: rcl_interfaces/srv/DescribeParameters
         /talker/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
         /talker/get_parameters: rcl_interfaces/srv/GetParameters
         /talker/list_parameters: rcl_interfaces/srv/ListParameters
         /talker/set_parameters: rcl_interfaces/srv/SetParameters
         /talker/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
       Service Clients:
       
       Action Servers:
       
       Action Clients:
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Launch a node
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 node list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /listener
       /talker
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-06:
On a default Windows 11 binary instalation with fastDDS.

Checked the 2 CLI commands for `ros2 node` as proposed.

All Ok, matching expectations.

[image]

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette