osrf/ros2_test_cases #523: Check ROS 2 Param CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check delete**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param delete /turtlesim background_b
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Deleting parameter failed: cannot undeclare a statically typed parameter
       ```
  </details>
- [ ] **Check describe**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param describe /turtlesim background_b
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Parameter name: background_b
       Type: integer
       Description: Blue channel of the background color
       Constraints:
         Min value: 0
         Max value: 255
         Step: 1
       ```
  </details>
- [ ] **Check dump**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param dump /turtlesim
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       /turtlesim:
         ros__parameters:
           background_b: 255
           background_g: 86
           background_r: 69
           qos_overrides:
             /parameter_events:
               publisher:
                 depth: 1000
                 durability: volatile
                 history: keep_last
                 reliability: reliable
           use_sim_time: false
       ```
  </details>
- [ ] **Check get**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param get /turtlesim background_b
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       Integer value is: 255
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param list /turtlesim
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       ros2 param list /turtlesim
       background_b
       background_g
       background_r
       qos_overrides./parameter_events.publisher.depth
       qos_overrides./parameter_events.publisher.durability
       qos_overrides./parameter_events.publisher.history
       qos_overrides./parameter_events.publisher.reliability
       use_sim_time
       ```
  </details>
- [ ] **Check set**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param set /turtlesim background_b 0
       ```
  
    **Expect**
    
    1.
       The background in the turtlesim window should have become green
       ```bash
       # stdout in terminal 2
       Set parameter successful
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
**The param CLI is working as expected. All the checks have passed.**

clalancette, 2023-05-09:
Thank you for testing!

closed 2023-05-09 by clalancette