osrf/ros2_test_cases #526: Check ROS 2 Pkg CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check create**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg create my-pkg
       ```
  
    **Expect**
    
    1.
       Creates a folder named `my-pkg`, which has a `package.xml` file, `CMakeLists.txt` file, `src` folder, and `include` folder.
    
  </details>
- [ ] **Check executables**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg executables turtlesim
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       turtlesim draw_square
       turtlesim mimic
       turtlesim turtle_teleop_key
       turtlesim turtlesim_node
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg list
       ```
  
    **Expect**
    
    1.
       Lists all the packages, for example.
       ```bash
       # stdout in terminal 1
       action_msgs
       action_tutorials_cpp
       action_tutorials_interfaces
       ...
       visualization_msgs
       yaml_cpp_vendor
       zstd_vendor
       ```
  </details>
- [ ] **Check prefix**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg prefix turtlesim
       ```
  
    **Expect**
    
    1.
       This varies depending on your install type. For example, from binaries on Ubuntu, it will be `/opt/ros/humble`
    
  </details>
- [ ] **Check xml**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg xml turtlesim
       ```
  
    **Expect**
    
    1.
       The full `package.xml` of the `turtlesim` package is printed to stdout.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
**All the pkg CLI runs as expected.**

Yadunund, 2023-05-09:
Thanks for testing!

closed 2023-05-09 by Yadunund