osrf/ros2_test_cases #529: Check ROS 2 Pkg CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check create**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg create my-pkg
       ```
  
    **Expect**
    
    1.
       Creates a folder named `my-pkg`, which has a `package.xml` file, `CMakeLists.txt` file, `src` folder, and `include` folder.
    
  </details>
- [ ] **Check executables**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg executables turtlesim
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       turtlesim draw_square
       turtlesim mimic
       turtlesim turtle_teleop_key
       turtlesim turtlesim_node
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg list
       ```
  
    **Expect**
    
    1.
       Lists all the packages, for example.
       ```bash
       # stdout in terminal 1
       action_msgs
       action_tutorials_cpp
       action_tutorials_interfaces
       ...
       visualization_msgs
       yaml_cpp_vendor
       zstd_vendor
       ```
  </details>
- [ ] **Check prefix**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg prefix turtlesim
       ```
  
    **Expect**
    
    1.
       This varies depending on your install type. For example, from binaries on Ubuntu, it will be `/opt/ros/humble`
    
  </details>
- [ ] **Check xml**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg xml turtlesim
       ```
  
    **Expect**
    
    1.
       The full `package.xml` of the `turtlesim` package is printed to stdout.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
- create
```
❯ ros2 pkg create my-pkg
going to create a new package
package name: my-pkg
destination directory: /tmp
package format: 3
version: 0.0.0
description: TODO: Package description
maintainer: ['daisuke <daisuke.nishimatsu1021@gmail.com>']
licenses: ['TODO: License declaration']
build type: ament_cmake
dependencies: []
creating folder ./my-pkg
creating ./my-pkg/package.xml
creating source and include folder
creating folder ./my-pkg/src
creating folder ./my-pkg/include/my-pkg
creating ./my-pkg/CMakeLists.txt

[WARNING]: Unknown license 'TODO: License declaration'.  This has been set in the package.xml, but no LICENSE file has been created.
It is recommended to use one of the ament license identitifers:
Apache-2.0
BSL-1.0
BSD-2.0
BSD-2-Clause
BSD-3-Clause
GPL-3.0-only
LGPL-3.0-only
MIT
MIT-0

/tmp
❯ ls my-pkg
CMakeLists.txt  include  package.xml  src
```

wep21, 2023-05-06:
- executables
```
❯ ros2 pkg executables turtlesim 
turtlesim draw_square
turtlesim mimic
turtlesim turtle_teleop_key
turtlesim turtlesim_node
```
- list
```
❯ ros2 pkg list
action_msgs
action_tutorials_cpp
action_tutorials_interfaces
ament_cmake
ament_cmake_core
ament_cmake_export_definitions
ament_cmake_export_dependencies
ament_cmake_export_include_directories
ament_cmake_export_interfaces
ament_cmake_export_libraries
ament_cmake_export_link_flags
ament_cmake_export_targets
ament_cmake_gen_version_h
ament_cmake_gmock
ament_cmake_gtest
ament_cmake_include_directories
ament_cmake_libraries
ament_cmake_pytest
ament_cmake_python
ament_cmake_ros
ament_cmake_target_dependencies
ament_cmake_test
ament_cmake_version
ament_copyright
ament_index_cpp
ament_index_python
ament_lint
ament_package
builtin_interfaces
class_loader
composition_interfaces
console_bridge_vendor
demo_nodes_cpp
demo_nodes_py
diagnostic_msgs
domain_coordinator
eigen3_cmake_module
example_interfaces
fastrtps_cmake_module
geometry_msgs
keyboard_handler
launch
launch_ros
launch_xml
launch_yaml
libstatistics_collector
libyaml_vendor
lifecycle_msgs
lifecycle_py
mcap_vendor
message_filters
nav_msgs
orocos_kdl_vendor
osrf_pycommon
pcl_conversions
pcl_msgs
pcl_ros
plotjuggler
plotjuggler_msgs
plotjuggler_ros
pluginlib
pybind11_vendor
python_cmake_module
rcl
rcl_action
rcl_interfaces
rcl_lifecycle
rcl_logging_interface
rcl_logging_spdlog
rcl_yaml_param_parser
rclcpp
rclcpp_action
rclcpp_components
rclpy
rcpputils
rcutils
rmw
rmw_cyclonedds_cpp
rmw_dds_common
rmw_implementation
rmw_implementation_cmake
ros2action
ros2bag
ros2cli
ros2cli_common_extensions
ros2component
ros2doctor
ros2interface
ros2launch
ros2lifecycle
ros2multicast
ros2node
ros2param
ros2pkg
ros2run
ros2service
ros2topic
ros_environment
ros_workspace
rosbag2
rosbag2_compression
rosbag2_compression_zstd
rosbag2_cpp
rosbag2_interfaces
rosbag2_py
rosbag2_storage
rosbag2_storage_default_plugins
rosbag2_storage_mcap
rosbag2_storage_sqlite3
rosbag2_transport
rosgraph_msgs
rosidl_adapter
rosidl_cli
rosidl_core_runtime
rosidl_default_runtime
rosidl_dynamic_typesupport
rosidl_generator_c
rosidl_generator_cpp
rosidl_generator_py
rosidl_generator_type_description
rosidl_parser
rosidl_pycommon
rosidl_runtime_c
rosidl_runtime_cpp
rosidl_runtime_py
rosidl_typesupport_c
rosidl_typesupport_cpp
rosidl_typesupport_fastrtps_c
rosidl_typesupport_fastrtps_cpp
rosidl_typesupport_interface
rosidl_typesupport_introspection_c
rosidl_typesupport_introspection_cpp
rpyutils
sensor_msgs
service_msgs
shared_queues_vendor
spdlog_vendor
sqlite3_vendor
sros2
statistics_msgs
std_msgs
std_srvs
test_msgs
tf2
tf2_geometry_msgs
tf2_msgs
tf2_py
tf2_ros
tf2_ros_py
tinyxml2_vendor
tracetools
turtlesim
type_description_interfaces
unique_identifier_msgs
yaml_cpp_vendor
zstd_vendor
```
- prefix
```
❯ ros2 pkg prefix turtlesim
/opt/ros/iron
```
- xml
```
❯ ros2 pkg xml turtlesim 
<package format="3">
  <name>turtlesim</name>
  <version>1.6.0</version>
  <description>
    turtlesim is a tool made for teaching ROS and ROS packages.
  </description>

  <maintainer email="alejandro@openrobotics.org">Alejandro Hernandez Cordero</maintainer>

  <license>BSD</license>

  <url type="website">http://www.ros.org/wiki/turtlesim</url>
  <url type="bugtracker">https://github.com/ros/ros_tutorials/issues</url>
  <url type="repository">https://github.com/ros/ros_tutorials</url>

  <author email="audrow@openrobotics.org">Audrow Nash</author>
  <author email="dthomas@osrfoundation.org">Dirk Thomas</author>
  <author>Josh Faust</author>
  <author email="mabel@openrobotics.org">Mabel Zhang</author>
  <author email="michael.jeronimo@openrobotics.org">Michael Jeronimo</author>
  <author email="sloretz@openrobotics.org">Shane Loretz</author>

  <build_depend>qt5-qmake</build_depend>
  <build_depend>qtbase5-dev</build_depend>

  <buildtool_depend>ament_cmake</buildtool_depend>
  <buildtool_depend>rosidl_default_generators</buildtool_depend>

  <exec_depend>libqt5-core</exec_depend>
  <exec_depend>libqt5-gui</exec_depend>
  <exec_depend>rosidl_default_runtime</exec_depend>

  <depend>ament_index_cpp</depend>
  <depend>geometry_msgs</depend>
  <depend>rclcpp</depend>
  <depend>rclcpp_action</depend>
  <depend>std_msgs</depend>
  <depend>std_srvs</depend>

  <member_of_group>rosidl_interface_packages</member_of_group>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette