osrf/ros2_test_cases #530: Check ROS 2 run CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check run**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You'll have to kill this command with Ctrl-C.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [INFO] [1649123362.906256690] [talker]: Publishing: 'Hello World: 1'
       [INFO] [1649123363.906469087] [talker]: Publishing: 'Hello World: 2'
       [INFO] [1649123364.905960807] [talker]: Publishing: 'Hello World: 3'
       ...
       ```
  </details>
- [ ] **Check run with prefix**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run --prefix 'gdb -ex run' demo_nodes_cpp talker
       ```
  
    **Expect**
    
    1.
       This runs the talker node in gdb. Once the talker is running, you can hit Ctrl-C to get to the console and then type `q` to quit gdb.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Yadunund, 2023-05-09:
`gdb` is not natively present on windows so skipping the second check.
cc: @mjcarroll

closed 2023-05-09 by Yadunund

clalancette, 2023-05-09:
> `gdb` is not natively present on windows so skipping the second check.

We should probably file a bug on https://github.com/audrow/yatm .  In particular, while gdb isn't present on Windows, we do want to check that we can run with a prefix, so maybe we should come up with a separate Windows-only test.

Yadunund, 2023-05-10:
Good call. https://github.com/audrow/yatm/issues/366