osrf/ros2_test_cases #532: Check ROS 2 run CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check run**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You'll have to kill this command with Ctrl-C.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [INFO] [1649123362.906256690] [talker]: Publishing: 'Hello World: 1'
       [INFO] [1649123363.906469087] [talker]: Publishing: 'Hello World: 2'
       [INFO] [1649123364.905960807] [talker]: Publishing: 'Hello World: 3'
       ...
       ```
  </details>
- [ ] **Check run with prefix**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run --prefix 'gdb -ex run' demo_nodes_cpp talker
       ```
  
    **Expect**
    
    1.
       This runs the talker node in gdb. Once the talker is running, you can hit Ctrl-C to get to the console and then type `q` to quit gdb.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

bai-ivan, 2023-05-06:
working on this, gonna grab gdb for windows

bai-ivan, 2023-05-06:
1) works
2) can't attach gdb to proccess under windows 

3)? compiled the talker from source and it works, if there's relevant issue you can mark it as done as well.

Yadunund, 2023-05-08:
Thanks for testing!

closed 2023-05-08 by Yadunund