osrf/ros2_test_cases #533: Check ROS 2 run CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check run**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You'll have to kill this command with Ctrl-C.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [INFO] [1649123362.906256690] [talker]: Publishing: 'Hello World: 1'
       [INFO] [1649123363.906469087] [talker]: Publishing: 'Hello World: 2'
       [INFO] [1649123364.905960807] [talker]: Publishing: 'Hello World: 3'
       ...
       ```
  </details>
- [ ] **Check run with prefix**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run --prefix 'gdb -ex run' demo_nodes_cpp talker
       ```
  
    **Expect**
    
    1.
       This runs the talker node in gdb. Once the talker is running, you can hit Ctrl-C to get to the console and then type `q` to quit gdb.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
- run
```
❯ ros2 run demo_nodes_cpp talker
[INFO] [1683348917.688293998] [talker]: Publishing: 'Hello World: 1'
[INFO] [1683348918.688248800] [talker]: Publishing: 'Hello World: 2'
[INFO] [1683348919.688230461] [talker]: Publishing: 'Hello World: 3'
```
- run with prefix
```
❯ ros2 run --prefix 'gdb -ex run' demo_nodes_cpp talker
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/iron/lib/demo_nodes_cpp/talker...
(No debugging symbols found in /opt/ros/iron/lib/demo_nodes_cpp/talker)
Starting program: /opt/ros/iron/lib/demo_nodes_cpp/talker 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff73ff640 (LWP 27093)]
[New Thread 0x7ffff6bfe640 (LWP 27094)]
[New Thread 0x7ffff5b92640 (LWP 27095)]
[New Thread 0x7ffff5391640 (LWP 27096)]
[New Thread 0x7ffff4ac9640 (LWP 27097)]
[New Thread 0x7fffeffff640 (LWP 27098)]
[New Thread 0x7fffef7fe640 (LWP 27099)]
[New Thread 0x7fffeeffd640 (LWP 27100)]
[New Thread 0x7fffee7fc640 (LWP 27101)]
[New Thread 0x7fffedffb640 (LWP 27102)]
[New Thread 0x7fffed7fa640 (LWP 27103)]
[INFO] [1683348974.174688927] [talker]: Publishing: 'Hello World: 1'
[INFO] [1683348975.174638890] [talker]: Publishing: 'Hello World: 2'
[INFO] [1683348976.174621661] [talker]: Publishing: 'Hello World: 3'
[INFO] [1683348977.174641105] [talker]: Publishing: 'Hello World: 4'
[INFO] [1683348978.174677113] [talker]: Publishing: 'Hello World: 5'
[INFO] [1683348979.174643665] [talker]: Publishing: 'Hello World: 6'
^C
Thread 1 "talker" received signal SIGINT, Interrupt.
__futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x7fffffffbc60, op=393, expected=0, futex_word=0x5555555ad398) at ./nptl/futex-internal.c:57
57	./nptl/futex-internal.c: No such file or directory.
(gdb) q
A debugging session is active.

	Inferior 1 [process 27090] will be killed.

Quit anyway? (y or n) y
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette