osrf/ros2_test_cases #536: Check ROS 2 security CLI on Windows

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Tutorial URL](https://github.com/ros2/sros2/blob/master/SROS2_Windows.md)

## Checks
- [ ] **The Linux tutorials work**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
The tutorial heading says to test on **FAST-RTPS only**

Yadunund, 2023-05-09:
The title says `Testing on Windows 10 (Tested with Fast-RTPS only)` which to me implies that it hasn't been tested with other RMWs. Not necessarily that it will not work with other RMWs (the readme references rmw_connextdds)

Having said that I also encountered the same issue as here https://github.com/osrf/ros2_test_cases/issues/537#issuecomment-1537265546

pradyum, 2023-05-09:
Hi I was able to run it with cyclone dds after downgrading the python module "cryptography==38.0.4". It is definitely a bug of version incompatibility between cryptography and pyopenssl modules. I saw the issue opened for https://github.com/ros2/sros2/issues/285.

[image]

Yadunund, 2023-05-09:
wow nice find! Will continue the discussion in the ticket.

closed 2023-05-09 by Yadunund