osrf/ros2_test_cases #537: Check ROS 2 security CLI on Windows

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Tutorial URL](https://github.com/ros2/sros2/blob/master/SROS2_Windows.md)

## Checks
- [ ] **The Linux tutorials work**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

bai-ivan, 2023-05-06:
working on this

bai-ivan, 2023-05-07:
Working on this.

It requres Administrator on Windows.

Following the tutorial letter by letter got this. 
```
C:\dev\ros2\sros2_demo>ros2 security create_keystore demo_keystore
Traceback (most recent call last):
  File "C:\dev\ros2_iron\Scripts\ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.25.0', 'console_scripts', 'ros2')())
  File "C:\dev\ros2_iron\Lib\site-packages\ros2cli\cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2_iron\Lib\site-packages\sros2\command\security.py", line 36, in main
    return extension.main(args=args)
  File "C:\dev\ros2_iron\Lib\site-packages\sros2\verb\create_keystore.py", line 38, in main
    sros2.keystore.create_keystore(args.ROOT)
  File "C:\dev\ros2_iron\Lib\site-packages\sros2\keystore\_keystore.py", line 82, in create_keystore
    _utilities.create_smime_signed_file(
  File "C:\dev\ros2_iron\Lib\site-packages\sros2\_utilities.py", line 67, in create_smime_signed_file
    f.write(_sign_bytes(cert, private_key, content))
  File "C:\dev\ros2_iron\Lib\site-packages\sros2\_utilities.py", line 140, in _sign_bytes
    flags = SSLBinding.lib.PKCS7_DETACHED
AttributeError: module 'lib' has no attribute 'PKCS7_DETACHED'
```

Yadunund, 2023-05-08:
To clarify, do you get this error when running without Admin privileges? Or even with?

bai-ivan, 2023-05-08:
I get this as Administrator. Without administrative access it complains about permissions.

Yadunund, 2023-05-09:
Gotcha. I've opened this ticket upstream https://github.com/ros2/sros2/issues/285

Yadunund, 2023-05-18:
Closing this since we have a bug ticketed.