osrf/ros2_test_cases #539: Check ROS 2 service CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check call**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_server
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 service call /add_two_ints example_interfaces/srv/AddTwoInts "{a: 1, b: 2}"
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [INFO] [1649123971.736926946] [add_two_ints_server]: Incoming request
       a: 1 b: 2
       ```
    1.
       ```bash
       # stdout in terminal 2
       requester: making request: example_interfaces.srv.AddTwoInts_Request(a=1, b=2)
       
       response:
       example_interfaces.srv.AddTwoInts_Response(sum=3)
       ```
  </details>
- [ ] **Check find**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_server
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 service find example_interfaces/srv/AddTwoInts
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /add_two_ints
       ```
    1.
       Trying to find another service message that is not being used will return no text to stdout; for example `ros2 service find example_interfaces/srv/SetBool` returns nothing.
    
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_server
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 service list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /add_two_ints
       /add_two_ints_server/describe_parameters
       /add_two_ints_server/get_parameter_types
       /add_two_ints_server/get_parameters
       /add_two_ints_server/list_parameters
       /add_two_ints_server/set_parameters
       /add_two_ints_server/set_parameters_atomically
       ```
  </details>
- [ ] **Check type**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_server
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 service type /add_two_ints
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       example_interfaces/srv/AddTwoInts
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
**The service CLI is working as expected. All the checks have passed.**

clalancette, 2023-05-09:
Thank you for testing!

closed 2023-05-09 by clalancette