osrf/ros2_test_cases #542: Check ROS 2 topic CLI

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check bw**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic bw /joint_states
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       Subscribed to [/joint_states]
       5.04 KB/s from 50 messages
         Message size mean: 0.10 KB min: 0.10 KB max: 0.10 KB
       5.02 KB/s from 100 messages
         Message size mean: 0.10 KB min: 0.10 KB max: 0.10 KB
       ...
       ```
  </details>
- [ ] **Check delay**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic delay /joint_states
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       average delay: 0.001
         min: 0.000s max: 0.001s std dev: 0.00012s window: 47
       average delay: 0.001
         min: 0.000s max: 0.002s std dev: 0.00021s window: 97
       ...
       ```
  </details>
- [ ] **Check echo**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic echo /joint_states
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       ---
       header:
         stamp:
           sec: 1649125169
           nanosec: 961952254
         frame_id: ''
       name:
       - single_rrbot_joint1
       - single_rrbot_joint2
       position:
       - -0.9985924716759744
       - -0.9985924716759744
       velocity: []
       effort: []
       ---
       ...
       ```
  </details>
- [ ] **Check find**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic find sensor_msgs/msg/JointState
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /joint_states
       ```
  </details>
- [ ] **Check hz**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic hz /joint_states
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       average rate: 50.252
         min: 0.000s max: 0.043s std dev: 0.00451s window: 52
       average rate: 50.125
         min: 0.000s max: 0.043s std dev: 0.00492s window: 102
       ...
       ```
  </details>
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic info /joint_states
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Type: sensor_msgs/msg/JointState
       Publisher count: 1
       Subscription count: 1
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /joint_states
       /map
       /parameter_events
       /robot_description
       /rosout
       /scan
       /tf
       /tf_static
       ```
  </details>
- [ ] **Check type**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic type /joint_states
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       sensor_msgs/msg/JointState
       ```
  </details>
- [ ] **Check pub**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp listener
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic pub /chatter std_msgs/msg/String "{data: 'hi'}"
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 1
       [INFO] [1649125551.479505602] [listener]: I heard: [hi]
       [INFO] [1649125552.480104544] [listener]: I heard: [hi]
       ...
       ```
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       publisher: beginning loop
       publishing #1: std_msgs.msg.String(data='hi')
       
       publishing #2: std_msgs.msg.String(data='hi')
       ...
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
**All the topic CLI works as expected.**

Yadunund, 2023-05-09:
Thanks for testing!

closed 2023-05-09 by Yadunund