osrf/ros2_test_cases #546: Check ROS 2 wtf CLI

Yadunund, 2023-04-05:
`ros2 wtf` is an alias for `ros2 doctor`.

## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `-r`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -r
       ```
  
    **Expect**
    
    1.
       A large report is generated with several sections.
    
  </details>
- [ ] **Check `-rf`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -rf
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., 'All 5 checks passed').
    
  </details>
- [ ] **Check `-iw`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -iw
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., '2/5 checks passed').
    
  </details>
- [ ] **Check hello**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf hello
       ```
  
    **Expect**
    
    1.
       You'll see repeated messages that look somewhat like the following.
       ```bash
       # stdout in terminal 1
       MULTIMACHINE COMMUNICATION SUMMARY
       Topic: /canyouhearme, Published Msg Count: 11
       Subscribed from:
                       Hostname             Msg Count /1.0s
       Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
       Received from:
                       Hostname             Msg Count /1.0s
       ------------------------------------------------------------
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
everything works as expected except ```-iw``` in which network module failed

[image]

Yadunund, 2023-05-09:
> everything works as expected except `-iw` in which network module failed
> 
> [image]

I think this is expected given the first printout from the command
```
C:\ros2-windows\Lib\site-packages\ros2doctor\api\network.py: 154: UserWarning: Interface flags are not available on Windows. Run `ipconfig` to see what interfaces are available.
```

I'll close this ticket. Thanks for testing.