osrf/ros2_test_cases #548: Check ROS 2 wtf CLI

Yadunund, 2023-04-05:
`ros2 wtf` is an alias for `ros2 doctor`.

## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `-r`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -r
       ```
  
    **Expect**
    
    1.
       A large report is generated with several sections.
    
  </details>
- [ ] **Check `-rf`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -rf
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., 'All 5 checks passed').
    
  </details>
- [ ] **Check `-iw`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -iw
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., '2/5 checks passed').
    
  </details>
- [ ] **Check hello**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf hello
       ```
  
    **Expect**
    
    1.
       You'll see repeated messages that look somewhat like the following.
       ```bash
       # stdout in terminal 1
       MULTIMACHINE COMMUNICATION SUMMARY
       Topic: /canyouhearme, Published Msg Count: 11
       Subscribed from:
                       Hostname             Msg Count /1.0s
       Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
       Received from:
                       Hostname             Msg Count /1.0s
       ------------------------------------------------------------
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

mjcarroll, 2023-05-04:
```
PS C:\ros2_iron> ros2 wtf -r
C:\ros2_iron\Lib\site-packages\ros2doctor\api\__init__.py: 154: UserWarning: Fail to call PackageReport class functions.

   NETWORK CONFIGURATION
inet    : 169.254.154.112
inet4   : ['169.254.154.112']
inet6   : ['fe80::bcd7:f94a:9ca7:8d6d']
netmask : 255.255.0.0
device  : Ethernet
mtu     : 1500
inet    : 169.254.156.136
inet4   : ['169.254.156.136']
inet6   : ['fe80::33c6:6d0d:e384:a500']
netmask : 255.255.0.0
device  : Local Area Connection* 1
mtu     : 1500
inet    : 169.254.216.212
inet4   : ['169.254.216.212']
inet6   : ['fe80::ed41:4c5b:c0ff:3cb6']
netmask : 255.255.0.0
device  : Local Area Connection* 2
mtu     : 1500
inet    : 192.168.68.100
inet4   : ['192.168.68.100']
inet6   : ['fe80::e2c5:8354:ef51:5994']
netmask : 255.255.252.0
device  : Wi-Fi
mtu     : 1500
inet    : 169.254.18.15
inet4   : ['169.254.18.15']
inet6   : ['fe80::5769:a065:9fc0:2ccd']
netmask : 255.255.0.0
device  : Bluetooth Network Connection
mtu     : 1500
inet    : 172.25.112.1
inet4   : ['172.25.112.1']
inet6   : ['fe80::a77d:ce54:1e18:2b9f']
netmask : 255.255.240.0
device  : vEthernet (nat)
mtu     : 1500
inet    : 172.26.80.1
inet4   : ['172.26.80.1']
inet6   : ['fe80::3cf4:a5fe:7f06:fa35']
netmask : 255.255.240.0
device  : vEthernet (Default Switch)
mtu     : 1500
inet    : 127.0.0.1
inet4   : ['127.0.0.1']
inet6   : ['::1']
netmask : 255.0.0.0
device  : Loopback Pseudo-Interface 1
mtu     : 1500

   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.22000-SP0
release          : 10
processor        : Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

   QOS COMPATIBILITY LIST
topic [type]            : /parameter_events [rcl_interfaces/msg/ParameterEvent]
publisher node          : listener
subscriber node         : listener
compatibility status    : OK
topic [type]            : /parameter_events [rcl_interfaces/msg/ParameterEvent]
publisher node          : _ros2cli_daemon_0_404b9fc7c2544310bfa4fa2e49ce6ba5
subscriber node         : listener
compatibility status    : OK

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : rolling
distribution type      : ros2
distribution status    : rolling
release platforms      : {'debian': ['bullseye'], 'rhel': ['9'], 'ubuntu': ['jammy']}

   TOPIC LIST
topic               : /chatter
publisher count     : 0
subscriber count    : 1
```

mjcarroll, 2023-05-04:
@Yadunund and @clalancette do we expect the distribution to be "rolling" here?

clalancette, 2023-05-04:
> @Yadunund and @clalancette do we expect the distribution to be "rolling" here?

No, that is definitely a bug.

clalancette, 2023-05-04:
> No, that is definitely a bug.

Oh, I think I know what happened here.  These binary archives were probably built before we merged in https://github.com/ros/ros_environment/pull/38 , so it will still say `rolling`.  That should be resolved once we rebuild the binary archives for release.

mjcarroll, 2023-05-04:
`ROS_DISTRO` is set to `rolling` in my environment as well, which is probably where doctor is reading it from.

mjcarroll, 2023-05-04:
Ah, well that is consistent with the ros_environment theory, lets make a note to recheck once those are rebuilt.

Yadunund, 2023-05-05:
I've opened a discussion to follow up https://github.com/osrf/ros2_test_cases/discussions/1050

I guess we can consider this done for now!

mjcarroll, 2023-05-05:
Also, `ros2 wtf hello` is hanging when shutting down:

```
PS C:\ros2_iron> ros2 wtf hello
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 11
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Python38\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Python38\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\ros2_iron\Lib\site-packages\rclpy\executors.py", line 280, in spin
    self.spin_once()
  File "C:\ros2_iron\Lib\site-packages\rclpy\executors.py", line 717, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "C:\ros2_iron\Lib\site-packages\rclpy\executors.py", line 703, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "C:\ros2_iron\Lib\site-packages\rclpy\executors.py", line 604, in _wait_for_ready_callbacks
    raise ExternalShutdownException()
rclpy.executors.ExternalShutdownException
```

Yadunund, 2023-05-09:
I think this can be closed again following the ros2doctor fix

clalancette, 2023-06-23:
With the tutorial party over, I'm going to close this out.