osrf/ros2_test_cases #549: Check ROS 2 wtf CLI

Yadunund, 2023-04-05:
`ros2 wtf` is an alias for `ros2 doctor`.

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `-r`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -r
       ```
  
    **Expect**
    
    1.
       A large report is generated with several sections.
    
  </details>
- [ ] **Check `-rf`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -rf
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., 'All 5 checks passed').
    
  </details>
- [ ] **Check `-iw`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -iw
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., '2/5 checks passed').
    
  </details>
- [ ] **Check hello**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf hello
       ```
  
    **Expect**
    
    1.
       You'll see repeated messages that look somewhat like the following.
       ```bash
       # stdout in terminal 1
       MULTIMACHINE COMMUNICATION SUMMARY
       Topic: /canyouhearme, Published Msg Count: 11
       Subscribed from:
                       Hostname             Msg Count /1.0s
       Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
       Received from:
                       Hostname             Msg Count /1.0s
       ------------------------------------------------------------
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
- -r
```
❯ ros2 wtf -r

   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<LOOPBACK,UP,RUNNING>
mtu          : 65536
inet         : 192.168.3.4
inet4        : ['192.168.3.4']
ether        : 14:75:5b:c0:97:6b
inet6        : ['fe80::4bfb:cec8:fd78:8b8a%wlp9s0']
netmask      : 255.255.255.0
device       : wlp9s0
flags        : 4163<MULTICAST,BROADCAST,UP,RUNNING>
mtu          : 1500
broadcast    : 192.168.3.255
inet         : 172.19.0.1
inet4        : ['172.19.0.1']
ether        : 02:42:e4:d4:6e:b5
inet6        : ['fe80::42:e4ff:fed4:6eb5%br-2b82ddfe6d03']
netmask      : 255.255.0.0
device       : br-2b82ddfe6d03
flags        : 4163<MULTICAST,BROADCAST,UP,RUNNING>
mtu          : 1500
broadcast    : 172.19.255.255
inet         : 172.23.0.1
inet4        : ['172.23.0.1']
ether        : 02:42:df:e7:51:b7
inet6        : ['fc00:f853:ccd:e793::1', 'fe80::42:dfff:fee7:51b7%br-9d78998eea49', 'fe80::1%br-9d78998eea49']
netmask      : 255.255.0.0
device       : br-9d78998eea49
flags        : 4163<MULTICAST,BROADCAST,UP,RUNNING>
mtu          : 1500
broadcast    : 172.23.255.255
inet         : 172.17.0.1
inet4        : ['172.17.0.1']
ether        : 02:42:3a:3c:26:f4
netmask      : 255.255.0.0
device       : docker0
flags        : 4099<MULTICAST,BROADCAST,UP>
mtu          : 1500
broadcast    : 172.17.255.255
ether        : 9a:c1:78:e0:9c:73
inet6        : ['fe80::98c1:78ff:fee0:9c73%veth820891a']
device       : veth820891a
flags        : 4163<MULTICAST,BROADCAST,UP,RUNNING>
mtu          : 1500
ether        : 0e:22:b2:34:b4:57
inet6        : ['fe80::c22:b2ff:fe34:b457%veth2e1bfb1']
device       : veth2e1bfb1
flags        : 4163<MULTICAST,BROADCAST,UP,RUNNING>
mtu          : 1500
ether        : f6:56:5a:23:9a:68
inet6        : ['fe80::f456:5aff:fe23:9a68%veth1b708dc']
device       : veth1b708dc
flags        : 4163<MULTICAST,BROADCAST,UP,RUNNING>
mtu          : 1500
ether        : 82:3a:c6:a4:b9:2b
inet6        : ['fe80::803a:c6ff:fea4:b92b%veth435dee7']
device       : veth435dee7
flags        : 4163<MULTICAST,BROADCAST,UP,RUNNING>
mtu          : 1500
ether        : ee:57:05:37:38:d8
inet6        : ['fe80::ec57:5ff:fe37:38d8%veth6286d58']
device       : veth6286d58
[... 354 lines cut ...]
```
- -rf
```
❯ ros2 wtf -rf

All 5 checks passed
```
- -iw
```
❯ ros2 wtf -iw

All 5 checks passed
```
- hello
```
❯ ros2 wtf hello
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette