osrf/ros2_test_cases #570: Executables in `composition`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `linktime_composition`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run composition linktime_composition
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Yadunund, 2023-05-09:
@clalancette do we expect linktimec_composition to work on windows? 🤔

clalancette, 2023-05-09:
> @clalancette do we expect `linktime_composition` to work on windows? thinking

There is no explicit code to disable functionality on Windows.  So I would say yes, we generally expect it to work.

mjcarroll, 2023-05-10:
Linktime composition explicitly doesn't work on Windows, there is an open issue here: https://github.com/ros2/demos/issues/514

Yadunund, 2023-05-11:
Thanks for the info. I'll close this and have made a not to update yatm

closed 2023-05-11 by Yadunund