osrf/ros2_test_cases #573: Executables in `composition`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `linktime_composition`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run composition linktime_composition
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
```
❯ ros2 run composition linktime_composition
[INFO] [1683350002.098273007] [linktime_composition]: Load library 
[INFO] [1683350002.098312358] [linktime_composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<composition::Client>
[INFO] [1683350002.106180986] [linktime_composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<composition::Listener>
[INFO] [1683350002.107536966] [linktime_composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<composition::Server>
[INFO] [1683350002.108638791] [linktime_composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<composition::Talker>
[INFO] [1683350003.110288935] [talker]: Publishing: 'Hello World: 1'
[INFO] [1683350003.110604185] [listener]: I heard: [Hello World: 1]
[INFO] [1683350004.106724854] [Server]: Incoming request: [a: 2, b: 3]
[INFO] [1683350004.107040892] [Client]: Got result: [5]
[INFO] [1683350004.110167516] [talker]: Publishing: 'Hello World: 2'
[INFO] [1683350004.110338271] [listener]: I heard: [Hello World: 2]
[INFO] [1683350005.110154951] [talker]: Publishing: 'Hello World: 3'
[INFO] [1683350005.110272279] [listener]: I heard: [Hello World: 3]
^C[INFO] [1683350005.765424983] [rclcpp]: signal_handler(signum=2)
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette