osrf/ros2_test_cases #576: Executables in `demo_nodes_cpp_native`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp_native talker
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with fastDDS.

Checked 1 executable is present and working.

All Ok.

[image]

clalancette, 2023-05-05:
Thank you for testing!

closed 2023-05-05 by clalancette