osrf/ros2_test_cases #577: Executables in `demo_nodes_cpp_native`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp_native talker
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
```
❯ ros2 run demo_nodes_cpp_native talker
[INFO] [1683350075.319650173] [talker_native]: eprosima::fastdds::dds::DomainParticipant * 93862810579008
[INFO] [1683350075.320356820] [talker_native]: eprosima::fastdds::dds::DataWriter * 93862813340464
[INFO] [1683350075.819906130] [talker_native]: Publishing: 'Hello World: 1'
[INFO] [1683350076.319882444] [talker_native]: Publishing: 'Hello World: 2'
[INFO] [1683350076.819867833] [talker_native]: Publishing: 'Hello World: 3'
[INFO] [1683350077.319866349] [talker_native]: Publishing: 'Hello World: 4'
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette