osrf/ros2_test_cases #579: Executables in `demo_nodes_cpp`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
  
    
  </details>
- [ ] **Check `listener`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp listener
       ```
  
    
  </details>
- [ ] **Check `listener_best_effort`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp listener_best_effort
       ```
  
    
  </details>
- [ ] **Check `talker_serialized_message`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker_serialized_message
       ```
  
    
  </details>
- [ ] **Check `listener_serialized_message`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp listener_serialized_message
       ```
  
    
  </details>
- [ ] **Check `talker_loaned_message`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker_loaned_message
       ```
  
    
  </details>
- [ ] **Check `add_two_ints_server`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_server
       ```
  
    
  </details>
- [ ] **Check `add_two_ints_client`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_client
       ```
  
    
  </details>
- [ ] **Check `add_two_ints_client_async`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_client_async
       ```
  
    
  </details>
- [ ] **Check `list_parameters`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp list_parameters
       ```
  
    
  </details>
- [ ] **Check `list_parameters_async`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp list_parameters_async
       ```
  
    
  </details>
- [ ] **Check `set_and_get_parameters`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp set_and_get_parameters
       ```
  
    
  </details>
- [ ] **Check `set_and_get_parameters_async`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp set_and_get_parameters_async
       ```
  
    
  </details>
- [ ] **Check `parameter_events`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_events
       ```
  
    
  </details>
- [ ] **Check `parameter_events_async`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_events_async
       ```
  
    
  </details>
- [ ] **Check `parameter_blackboard`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_blackboard
       ```
  
    
  </details>
- [ ] **Check `even_parameters_node`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp even_parameters_node
       ```
  
    
  </details>
- [ ] **Check `one_off_timer`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp one_off_timer
       ```
  
    
  </details>
- [ ] **Check `reuse_timer`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp reuse_timer
       ```
  
    
  </details>
- [ ] **Check `allocator_tutorial`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp allocator_tutorial
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
All the node are executing and working as expected. The checks are passed.

clalancette, 2023-05-09:
Thank you for testing!

closed 2023-05-09 by clalancette