osrf/ros2_test_cases #581: Executables in `demo_nodes_cpp`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
  
    
  </details>
- [ ] **Check `listener`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp listener
       ```
  
    
  </details>
- [ ] **Check `listener_best_effort`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp listener_best_effort
       ```
  
    
  </details>
- [ ] **Check `talker_serialized_message`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker_serialized_message
       ```
  
    
  </details>
- [ ] **Check `listener_serialized_message`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp listener_serialized_message
       ```
  
    
  </details>
- [ ] **Check `talker_loaned_message`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker_loaned_message
       ```
  
    
  </details>
- [ ] **Check `add_two_ints_server`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_server
       ```
  
    
  </details>
- [ ] **Check `add_two_ints_client`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_client
       ```
  
    
  </details>
- [ ] **Check `add_two_ints_client_async`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_client_async
       ```
  
    
  </details>
- [ ] **Check `list_parameters`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp list_parameters
       ```
  
    
  </details>
- [ ] **Check `list_parameters_async`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp list_parameters_async
       ```
  
    
  </details>
- [ ] **Check `set_and_get_parameters`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp set_and_get_parameters
       ```
  
    
  </details>
- [ ] **Check `set_and_get_parameters_async`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp set_and_get_parameters_async
       ```
  
    
  </details>
- [ ] **Check `parameter_events`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_events
       ```
  
    
  </details>
- [ ] **Check `parameter_events_async`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_events_async
       ```
  
    
  </details>
- [ ] **Check `parameter_blackboard`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_blackboard
       ```
  
    
  </details>
- [ ] **Check `even_parameters_node`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp even_parameters_node
       ```
  
    
  </details>
- [ ] **Check `one_off_timer`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp one_off_timer
       ```
  
    
  </details>
- [ ] **Check `reuse_timer`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp reuse_timer
       ```
  
    
  </details>
- [ ] **Check `allocator_tutorial`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp allocator_tutorial
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
- talker
```
❯ ros2 run demo_nodes_cpp talker
[INFO] [1683350117.451682361] [talker]: Publishing: 'Hello World: 1'
[INFO] [1683350118.451663264] [talker]: Publishing: 'Hello World: 2'
[INFO] [1683350119.451688887] [talker]: Publishing: 'Hello World: 3'
```
- listener
```
❯ ros2 run demo_nodes_cpp listener
[INFO] [1683350168.496788135] [listener]: I heard: [Hello World: 1]
[INFO] [1683350169.496811706] [listener]: I heard: [Hello World: 2]
[INFO] [1683350170.496839985] [listener]: I heard: [Hello World: 3]
```
- listener best effort
```
❯ ros2 run demo_nodes_cpp listener_best_effort
[INFO] [1683350201.168230375] [listener]: I heard: [Hello World: 6]
[INFO] [1683350202.167971893] [listener]: I heard: [Hello World: 7]
[INFO] [1683350203.167905560] [listener]: I heard: [Hello World: 8]
[INFO] [1683350204.167930913] [listener]: I heard: [Hello World: 9]
^C[INFO] [1683350204.547857503] [rclcpp]: signal_handler(signum=2)
```

wep21, 2023-05-06:
- talker serialized message
```
❯ ros2 run demo_nodes_cpp talker_serialized_message 
ROS message:
Hello World:1
serialized message:
00 01 00 00 0e 00 00 00 48 65 6c 6c 6f 20 57 6f 72 6c 64 3a 31 00 
ROS message:
Hello World:2
serialized message:
00 01 00 00 0e 00 00 00 48 65 6c 6c 6f 20 57 6f 72 6c 64 3a 32 00 
ROS message:
Hello World:3
serialized message:
00 01 00 00 0e 00 00 00 48 65 6c 6c 6f 20 57 6f 72 6c 64 3a 33 00 
```
- listener_serialized_message
```
❯ ros2 run demo_nodes_cpp listener_serialized_message
I heard data of length: 24
00 01 00 00 0e 00 00 00 48 65 6c 6c 6f 20 57 6f 72 6c 64 3a 32 00 00 00 
serialized data after deserialization: Hello World:2
I heard data of length: 24
00 01 00 00 0e 00 00 00 48 65 6c 6c 6f 20 57 6f 72 6c 64 3a 33 00 00 00 
serialized data after deserialization: Hello World:3
I heard data of length: 24
00 01 00 00 0e 00 00 00 48 65 6c 6c 6f 20 57 6f 72 6c 64 3a 34 00 00 00 
serialized data after deserialization: Hello World:4
```
- talker loaned message
```
❯ ros2 run demo_nodes_cpp talker_loaned_message
[INFO] [1683350357.349182944] [loaned_message_talker]: Publishing: '1.000000'
[INFO] [1683350357.349305891] [rclcpp]: Currently used middleware can't loan messages. Local allocator will be used.
[INFO] [1683350357.349315360] [loaned_message_talker]: Publishing: 'Hello World: 1'
[INFO] [1683350358.349087998] [loaned_message_talker]: Publishing: '2.000000'
[INFO] [1683350358.349144126] [loaned_message_talker]: Publishing: 'Hello World: 2'
```

wep21, 2023-05-06:
- add_two_ints_server
```
❯ ros2 run demo_nodes_cpp add_two_ints_server 
[INFO] [1683350409.139137244] [add_two_ints_server]: Incoming request
a: 2 b: 3
```

-  add_two_ints_client
```
❯ ros2 run demo_nodes_cpp add_two_ints_client
[INFO] [1683350409.139617124] [add_two_ints_client]: Result of add_two_ints: 5
```

- add_two_ints_client_async
```
❯ ros2 run demo_nodes_cpp add_two_ints_client_async 
[INFO] [1683350516.295773912] [add_two_ints_client]: Result of add_two_ints: 5
```

wep21, 2023-05-06:
```
❯ ros2 run demo_nodes_cpp list_parameters
[INFO] [1683350557.532216137] [list_paramters]: Setting parameters...
[INFO] [1683350557.532442473] [list_paramters]: Listing parameters...
[INFO] [1683350557.532516840] [list_paramters]: 
Parameter names:
 bar
 foo
 foo.first
 foo.second
Parameter prefixes:
 foo
```
```
❯ ros2 run demo_nodes_cpp list_parameters_async
[INFO] [1683350566.829452837] [list_parameters_async]: Setting parameters...
[INFO] [1683350566.829755295] [list_parameters_async]: Listing parameters...
[INFO] [1683350566.829834866] [list_parameters_async]: 
Parameter names:
 bar
 foo
 foo.first
 foo.second
Parameter prefixes:
 foo
```
```
❯ ros2 run demo_nodes_cpp set_and_get_parameters
[INFO] [1683350575.684991094] [set_and_get_parameters]: 
Parameter name: foo
Parameter value (integer): 2
Parameter name: baz
Parameter value (double): 1.450000
Parameter name: foobarbaz
Parameter value (bool_array): [true, false]
Parameter name: toto
Parameter value (byte_array): [0xff, 0x7f]
```
```
❯ ros2 run demo_nodes_cpp set_and_get_parameters_async
[INFO] [1683350581.234907543] [set_and_get_parameters_async]: 
Parameter name: foo
Parameter value (integer): 2
Parameter name: baz
Parameter value (double): 1.450000
Parameter name: foobarbaz
Parameter value (bool_array): [true, false]
Parameter name: toto
Parameter value (byte_array): [0xff, 0x7f]
```

wep21, 2023-05-06:
- parameter events
```
❯ ros2 run demo_nodes_cpp parameter_events 
[INFO] [1683350710.320728245] [parameter_events]: 
Parameter event:
 new parameters:
  foo
 changed parameters:
 deleted parameters:

[INFO] [1683350710.320864435] [parameter_events]: 
Parameter event:
 new parameters:
  bar
 changed parameters:
 deleted parameters:

[INFO] [1683350710.320912527] [parameter_events]: 
Parameter event:
 new parameters:
  baz
 changed parameters:
 deleted parameters:

[INFO] [1683350710.320944968] [parameter_events]: 
Parameter event:
 new parameters:
  foobar
 changed parameters:
 deleted parameters:

[INFO] [1683350710.320971348] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
  foo
 deleted parameters:

[INFO] [1683350710.320985574] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
  bar
 deleted parameters:

[INFO] [1683350710.320998669] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
  baz
 deleted parameters:

[INFO] [1683350710.321011004] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
  foobar
 deleted parameters:

[INFO] [1683350710.321024741] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
  foo
 deleted parameters:

[INFO] [1683350710.321042076] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
  bar
 deleted parameters:
```
- parameter events async
```
❯ ros2 run demo_nodes_cpp parameter_events_async 
[INFO] [1683350719.218864715] [parameter_events]: 
Parameter event:
 new parameters:
  foo
 changed parameters:
 deleted parameters:

[INFO] [1683350719.218935759] [parameter_events]: 
Parameter event:
 new parameters:
  bar
 changed parameters:
 deleted parameters:

[INFO] [1683350719.218951745] [parameter_events]: 
Parameter event:
 new parameters:
  baz
 changed parameters:
 deleted parameters:

[INFO] [1683350719.218965678] [parameter_events]: 
Parameter event:
 new parameters:
  foobar
 changed parameters:
 deleted parameters:

[INFO] [1683350719.419081868] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
  foo
 deleted parameters:

[INFO] [1683350719.419162116] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
  bar
 deleted parameters:

[INFO] [1683350719.419192302] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
  baz
 deleted parameters:

[INFO] [1683350719.419228160] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
  foobar
 deleted parameters:

[INFO] [1683350719.419240928] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
  foo
 deleted parameters:

[INFO] [1683350719.419254619] [parameter_events]: 
Parameter event:
 new parameters:
 changed parameters:
  bar
 deleted parameters:
```
- parameter blackboard
```
❯ ros2 run demo_nodes_cpp parameter_blackboard
[INFO] [1683350728.065828631] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '5' parameters already!
```

wep21, 2023-05-06:
- even parameter node
```
❯ ros2 run demo_nodes_cpp even_parameters_node
[INFO] [1683350853.904895528] [even_parameters_node]: This example node shows a parameter callback that rejects
[INFO] [1683350853.904951066] [even_parameters_node]: all parameter updates except for those that set an even integer.
[INFO] [1683350853.904957071] [even_parameters_node]: Try running 'ros2 param set /even_parameters_node myint 2' to
[INFO] [1683350853.904961330] [even_parameters_node]: successfully set a parameter.
```
- one off timer
```
❯ ros2 run demo_nodes_cpp one_off_timer
[INFO] [1683350907.562180751] [one_off_timer]: in periodic_timer callback
[INFO] [1683350907.562327147] [one_off_timer]:   resetting one off timer
[INFO] [1683350908.562502001] [one_off_timer]: in one_off_timer callback
[INFO] [1683350909.562123207] [one_off_timer]: in periodic_timer callback
[INFO] [1683350909.562176660] [one_off_timer]:   not resetting one off timer
```
- reuse timer
```
❯ ros2 run demo_nodes_cpp reuse_timer
[INFO] [1683350941.440716983] [reuse_timer]: in periodic_timer callback
[INFO] [1683350941.440837680] [reuse_timer]:   resetting one off timer
[INFO] [1683350942.441047590] [reuse_timer]: in one_off_timer callback
[INFO] [1683350943.440691852] [reuse_timer]: in periodic_timer callback
[INFO] [1683350943.440818419] [reuse_timer]:   not resetting one off timer
```
- allocator tutorial
```
❯ ros2 run demo_nodes_cpp allocator_tutorial
This simple demo shows off a custom memory allocator to count all
instances of new/delete in the program.  It can be run in either regular
mode (no arguments), or in intra-process mode (by passing 'intra' as a
command-line argument).  It will then publish a message to the
'/allocator_tutorial' topic every 10 milliseconds until Ctrl-C is pressed.
At that time it will print a count of the number of allocations and
deallocations that happened during the program.

Intra-process pipeline is OFF.
^C[INFO] [1683350986.164292111] [rclcpp]: signal_handler(signum=2)
Global new was called 3795 times during spin
Global delete was called 3775 times during spin
Allocator new was called 8388 times during spin
Allocator delete was called 8369 times during spin
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette