osrf/ros2_test_cases #588: Executables in `dummy_robot_bringup`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `dummy_robot_bringup.launch.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
  
    
  </details>
- [ ] **Check `dummy_robot_bringup_launch.xml`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.xml
       ```
  
    
  </details>
- [ ] **Check `dummy_robot_bringup_launch.yaml`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.yaml
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with fastDDS.

ERROR: launching with `.xml` or `.yaml` files does not seem to work.
Launch files are properly present in binary installation, but only launching with the `.py` is working .

Error for two other option reports:
```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): file not found: [WinError 2] The system cannot find the file specified
```

I don't expect this to be specific to the `dummy_robot_bringup`.
I checked this is properly working on a `jammy` binary installation.

[image]

Yadunund, 2023-05-08:
Thanks for highlighting this. I've added a ticket upstream https://github.com/ros2/demos/issues/613

Yadunund, 2023-05-18:
We have an open PR for this fix so I'll close this ticket. https://github.com/ros2/launch/pull/708

closed 2023-05-18 by Yadunund