osrf/ros2_test_cases #589: Executables in `dummy_robot_bringup`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `dummy_robot_bringup.launch.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
  
    
  </details>
- [ ] **Check `dummy_robot_bringup_launch.xml`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.xml
       ```
  
    
  </details>
- [ ] **Check `dummy_robot_bringup_launch.yaml`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.yaml
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
- py
```
❯ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /home/daisuke/.ros/log/2023-05-06-17-53-45-775454-npc2301008-67530
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [67533]
[INFO] [robot_state_publisher-2]: process started with pid [67535]
[INFO] [dummy_joint_states-3]: process started with pid [67537]
[INFO] [dummy_laser-4]: process started with pid [67539]
[dummy_laser-4] [INFO] [1683363225.935474930] [dummy_laser]: angle inc:	0.004363
[dummy_laser-4] [INFO] [1683363225.935539437] [dummy_laser]: scan size:	1081
[dummy_laser-4] [INFO] [1683363225.935547812] [dummy_laser]: scan time increment: 	0.000028
[robot_state_publisher-2] [INFO] [1683363225.944944684] [robot_state_publisher]: got segment single_rrbot_camera_link
[robot_state_publisher-2] [INFO] [1683363225.944998822] [robot_state_publisher]: got segment single_rrbot_hokuyo_link
[robot_state_publisher-2] [INFO] [1683363225.945003583] [robot_state_publisher]: got segment single_rrbot_link1
[robot_state_publisher-2] [INFO] [1683363225.945006503] [robot_state_publisher]: got segment single_rrbot_link2
[robot_state_publisher-2] [INFO] [1683363225.945009208] [robot_state_publisher]: got segment single_rrbot_link3
[robot_state_publisher-2] [INFO] [1683363225.945011709] [robot_state_publisher]: got segment world
```
- xml
```
❯ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.xml 
[INFO] [launch]: All log files can be found below /home/daisuke/.ros/log/2023-05-06-17-54-18-317818-npc2301008-67712
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [67716]
[INFO] [robot_state_publisher-2]: process started with pid [67718]
[INFO] [dummy_joint_states-3]: process started with pid [67720]
[INFO] [dummy_laser-4]: process started with pid [67722]
[dummy_laser-4] [INFO] [1683363258.500213035] [dummy_laser]: angle inc:	0.004363
[dummy_laser-4] [INFO] [1683363258.500281309] [dummy_laser]: scan size:	1081
[dummy_laser-4] [INFO] [1683363258.500287388] [dummy_laser]: scan time increment: 	0.000028
[robot_state_publisher-2] [INFO] [1683363258.503071162] [robot_state_publisher]: got segment single_rrbot_camera_link
[robot_state_publisher-2] [INFO] [1683363258.503119139] [robot_state_publisher]: got segment single_rrbot_hokuyo_link
[robot_state_publisher-2] [INFO] [1683363258.503124014] [robot_state_publisher]: got segment single_rrbot_link1
[robot_state_publisher-2] [INFO] [1683363258.503126978] [robot_state_publisher]: got segment single_rrbot_link2
[robot_state_publisher-2] [INFO] [1683363258.503129575] [robot_state_publisher]: got segment single_rrbot_link3
[robot_state_publisher-2] [INFO] [1683363258.503132294] [robot_state_publisher]: got segment world
```
- yaml
```
❯ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.yaml
[INFO] [launch]: All log files can be found below /home/daisuke/.ros/log/2023-05-06-17-54-42-169756-npc2301008-67855
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [67859]
[INFO] [robot_state_publisher-2]: process started with pid [67861]
[INFO] [dummy_joint_states-3]: process started with pid [67863]
[INFO] [dummy_laser-4]: process started with pid [67865]
[dummy_laser-4] [INFO] [1683363282.350289480] [dummy_laser]: angle inc:	0.004363
[dummy_laser-4] [INFO] [1683363282.350346253] [dummy_laser]: scan size:	1081
[dummy_laser-4] [INFO] [1683363282.350353963] [dummy_laser]: scan time increment: 	0.000028
[robot_state_publisher-2] [INFO] [1683363282.352977441] [robot_state_publisher]: got segment single_rrbot_camera_link
[robot_state_publisher-2] [INFO] [1683363282.353027582] [robot_state_publisher]: got segment single_rrbot_hokuyo_link
[robot_state_publisher-2] [INFO] [1683363282.353032527] [robot_state_publisher]: got segment single_rrbot_link1
[robot_state_publisher-2] [INFO] [1683363282.353035516] [robot_state_publisher]: got segment single_rrbot_link2
[robot_state_publisher-2] [INFO] [1683363282.353038230] [robot_state_publisher]: got segment single_rrbot_link3
[robot_state_publisher-2] [INFO] [1683363282.353040701] [robot_state_publisher]: got segment world
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette