osrf/ros2_test_cases #593: Executables in `dummy_sensors`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `dummy_joint_states`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run dummy_sensors dummy_joint_states
       ```
  
    
  </details>
- [ ] **Check `dummy_laser`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run dummy_sensors dummy_laser
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

wep21, 2023-05-06:
- dummy joint states
```
❯ ros2 topic echo /joint_states --once
header:
  stamp:
    sec: 1683363361
    nanosec: 367613546
  frame_id: ''
name:
- single_rrbot_joint1
- single_rrbot_joint2
position:
- 0.9154525466207135
- 0.9154525466207135
velocity: []
effort: []
---
```
- dummy_laser
```
❯ ros2 run dummy_sensors dummy_laser
[INFO] [1683363407.173798802] [dummy_laser]: angle inc:	0.004363
[INFO] [1683363407.173870732] [dummy_laser]: scan size:	1081
[INFO] [1683363407.173876432] [dummy_laser]: scan time increment: 	0.000028
```

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette

kscottz, 2023-05-19:
@wep21 Hi Daisuke, you are one of our top issue closers for the test and tutorial party! Can I ask that you [fill out this short form](https://forms.gle/SDG2J3LJQTrtNAHH8) so we can send you some ROS Iron swag?

wep21, 2023-05-19:
@kscottz Thank you! I submitted the form you shared.