osrf/ros2_test_cases #595: Executables in `examples_rclcpp_minimal_action_client`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `action_client_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable
       ```
  
    
  </details>
- [ ] **Check `action_client_not_composable_with_cancel`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_cancel
       ```
  
    
  </details>
- [ ] **Check `action_client_not_composable_with_feedback`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_feedback
       ```
  
    
  </details>
- [ ] **Check `action_client_member_functions`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_member_functions
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
I first ran the example_rclcpp_minimal_action_server and tested these nodes. All the node have executed as expected. The checks are passed.

clalancette, 2023-05-09:
Thank you for testing!

closed 2023-05-09 by clalancette