osrf/ros2_test_cases #596: Executables in `examples_rclcpp_minimal_action_client`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `action_client_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable
       ```
  
    
  </details>
- [ ] **Check `action_client_not_composable_with_cancel`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_cancel
       ```
  
    
  </details>
- [ ] **Check `action_client_not_composable_with_feedback`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_not_composable_with_feedback
       ```
  
    
  </details>
- [ ] **Check `action_client_member_functions`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_action_client action_client_member_functions
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with fastDDS.

Check all 4 executables are present and luanchin ( and work with the server part).

All OK.

[image]

clalancette, 2023-05-05:
Thank you for testing!

closed 2023-05-05 by clalancette