osrf/ros2_test_cases #607: Executables in `examples_rclcpp_minimal_composition`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `composition_composed`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_composed
       ```
  
    
  </details>
- [ ] **Check `composition_publisher`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_publisher
       ```
  
    
  </details>
- [ ] **Check `composition_subscriber`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_subscriber
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
The nodes are executing and checks are passed.

clalancette, 2023-05-09:
Thank you for testing!

closed 2023-05-09 by clalancette