osrf/ros2_test_cases #608: Executables in `examples_rclcpp_minimal_composition`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `composition_composed`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_composed
       ```
  
    
  </details>
- [ ] **Check `composition_publisher`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_publisher
       ```
  
    
  </details>
- [ ] **Check `composition_subscriber`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_composition composition_subscriber
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with fast DDS.

Checked all 3 executables are present and launching.

All OK.

[image]

clalancette, 2023-05-05:
Thank you for testing!

closed 2023-05-05 by clalancette