osrf/ros2_test_cases #611: Executables in `examples_rclcpp_minimal_publisher`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `publisher_lambda`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_lambda
       ```
  
    
  </details>
- [ ] **Check `publisher_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_member_function
       ```
  
    
  </details>
- [ ] **Check `publisher_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_not_composable
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
The nodes are executing and tested with examples_rclcpp_minimal_subscriber package. The checks are passed.

clalancette, 2023-05-09:
Thank you for testing!

closed 2023-05-09 by clalancette