osrf/ros2_test_cases #612: Executables in `examples_rclcpp_minimal_publisher`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `publisher_lambda`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_lambda
       ```
  
    
  </details>
- [ ] **Check `publisher_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_member_function
       ```
  
    
  </details>
- [ ] **Check `publisher_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_publisher publisher_not_composable
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with fast DDS.

Checked all 3 executables are present and launching.

All OK.

[image]

clalancette, 2023-05-05:
Thank you for testing!

closed 2023-05-05 by clalancette