osrf/ros2_test_cases #621: Executables in `examples_rclcpp_minimal_subscriber`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `subscriber_lambda`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_subscriber subscriber_lambda
       ```
  
    
  </details>
- [ ] **Check `subscriber_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_subscriber subscriber_member_function
       ```
  
    
  </details>
- [ ] **Check `subscriber_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclcpp_minimal_subscriber subscriber_not_composable
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
All the nodes are executing when tested with examples_rclcpp_minimal_publisher package. All checks are passed.

clalancette, 2023-05-09:
Thank you for testing!

closed 2023-05-09 by clalancette