osrf/ros2_test_cases #636: Executables in `examples_rclpy_minimal_action_server`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `server_defer`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_defer
       ```
  
    
  </details>
- [ ] **Check `server_not_composable`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_not_composable
       ```
  
    
  </details>
- [ ] **Check `server`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server
       ```
  
    
  </details>
- [ ] **Check `server_queue_goals`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_queue_goals
       ```
  
    
  </details>
- [ ] **Check `server_single_goal`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_action_server server_single_goal
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with fastDDS.

Checked all 5 executables are present and launching.

All OK.

[image]

clalancette, 2023-05-05:
Thank you for testing!

closed 2023-05-05 by clalancette