osrf/ros2_test_cases #640: Executables in `examples_rclpy_minimal_client`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `client`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_client client
       ```
  
    
  </details>
- [ ] **Check `client_async`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_client client_async
       ```
  
    
  </details>
- [ ] **Check `client_async_member_function`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_client client_async_member_function
       ```
  
    
  </details>
- [ ] **Check `client_async_callback`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run examples_rclpy_minimal_client client_async_callback
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with fastDDS.

Checked all 4 executables are present and launching.

All Ok.

[image]

clalancette, 2023-05-05:
Thank you for testing!

closed 2023-05-05 by clalancette