osrf/ros2_test_cases #654: Executables in `gazebo_ros_pkgs`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `gazebo --verbose /opt/ros/rolling/share/gazebo_plugins/worlds/gazebo_ros_camera_distortion_barrel_demo.world`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run gazebo_ros_pkgs gazebo --verbose /opt/ros/rolling/share/gazebo_plugins/worlds/gazebo_ros_camera_distortion_barrel_demo.world
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-13:
Hello @Yadunund could you please assign this task to me?

juliamp22, 2023-05-13:
The package  is not installed in ROS2 windows. Do I have to install some package? Thank you so much. 
[image]

Yadunund, 2023-05-13:
gazebo is not officially supported on windows so this test case is not valid. I've taken note and will update the test generator to avoid this next time https://github.com/audrow/yatm/issues/366

closed 2023-05-13 by Yadunund

juliamp22, 2023-05-13:
Thank you so much @Yadunund !