osrf/ros2_test_cases #655: Executables in `gazebo_ros_pkgs`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `gazebo --verbose /opt/ros/rolling/share/gazebo_plugins/worlds/gazebo_ros_camera_distortion_barrel_demo.world`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run gazebo_ros_pkgs gazebo --verbose /opt/ros/rolling/share/gazebo_plugins/worlds/gazebo_ros_camera_distortion_barrel_demo.world
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
The gazebo_ros_pkgs is not installed and not found by default install with ```ros-iron-desktop-full```.

Yadunund, 2023-05-11:
It's not a meant to be part of `ros-iron-desktop`. Users will install it via `rosdep` or `apt`.

pradyum, 2023-05-11:
My bad, I installed the gazebo_ros_pkgs via ```apt```. But this is no executable in the gazebo_ros_pkgs, the ```gazebo``` executable is there as a standalone command. 

The successful execution used this command: ```gazebo --verbose /opt/ros/iron/share/gazebo_plugins/worlds/gazebo_ros_camera_distortion_barrel_demo.world```

The gazebo is executing and check is passed.

Yadunund, 2023-05-16:
Great, thanks for trying this out!

closed 2023-05-16 by Yadunund